Koenkk/zigbee2mqtt #25069: Sonoff SNZB-04P: Can't reset tamper

Anton2079, 2024-12-05:
### What happened?

Tamper detected by Zigbee device, visible on both Zigbee2Mqtt and HA.
Go to reset the tamper sensor on Zigbee2Mqtt and gets error:
`Z2m: Publish 'get' 'tamper' to 'Rear Bedroom Window' failed: 'Error: Cluster with name '#####' does not exist'`

### What did you expect to happen?

The state of tamper to change from 'Tampered' to 'Clear'

### How to reproduce it (minimal and precise)

I haven't been able to undo, so I can only assume trigger the tamper switch and then you'll be stuck in the same mode.

### Zigbee2MQTT version

1.42.0 commit: unknown

### Adapter firmware version

7.4.3 [GA]

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P / EmberZNet

### Setup

Intel NUC

### Debug log

`info 2024-12-05 19:12:06z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Rear BBQ Door Sensor', payload '{"battery":100,"battery_low":false,"contact":false,"last_seen":"2024-12-05T08:12:06.703Z","linkquality":104,"tamper":false,"update":{"installed_version":63,"latest_version":63,"state":"idle"},"update_available":null,"voltage":3100}'
error 2024-12-05 19:12:09z2m: Publish 'get' 'tamper' to 'Rear Bedroom Window' failed: 'Error: Cluster with name '64529' does not exist'`

Drac0fly, 2025-01-05:
Up please

francocm, 2025-01-19:
Same issue here

francocm, 2025-03-21:
Commenting to keep issue alive.

Starkstromkonsument, 2025-04-19:
I'm thinking about buying these sensors as a replacement for the [broken IKEA Parasoll](https://github.com/Koenkk/zigbee2mqtt/issues/22579) sensors. Is this issue likely to be fixed?

Starkstromkonsument, 2025-06-19:
Any news about this?

lrljoe, 2025-07-04:
Just seen this myself,I believe it's got an incorrect cluster reference, need to double check

```
    {
        zigbeeModel: ["SNZB-04P"],
        model: "SNZB-04P",
        vendor: "SONOFF",
        description: "Contact sensor",
        extend: [
            m.iasZoneAlarm({zoneType: "contact", zoneAttributes: ["alarm_1", "battery_low"]}),
            m.binary({
                name: "tamper",
                cluster: 0xfc11,
                attribute: {ID: 0x2000, type: 0x20},
                description: "Tamper-proof status",
                valueOn: [true, 0x01],
                valueOff: [false, 0x00],
                zigbeeCommandOptions: {manufacturerCode: Zcl.ManufacturerCode.SHENZHEN_COOLKIT_TECHNOLOGY_CO_LTD},
                access: "STATE_GET",
            }),
            ewelinkBattery(),
        ],
        ota: true,
    },
```