Koenkk/zigbee2mqtt #25070: [Wrong device]: eWeLink SNZB-05

RobertoCravallo, 2024-12-05:
### Link

https://www.zigbee2mqtt.io/devices/SNZB-05.html

### Model

SNZB-05

### Description

water sensor

### Vendor

eWeLink

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/SNZB-05.png

### Database entry

No idea how to get this data

### Notes

Was shown incorrectly already in 1.42, but correctly in 1.42.0-1. Now in 1.42.0-2 it is wrong again. So it broke with 1.42.0-2!!

It gets recognized as "snzb-03" instead of "snzb-05".

fat-freematilde, 2025-01-23:
I just created an issue and realised that is wat a duplicate of this one.

I made a very nasty workaround - I have the exact same issue on version z2m 2.0
https://github.com/Koenkk/zigbee2mqtt/issues/25954

RobertoCravallo, 2025-03-25:
Still shoving wrong, team does not seem to care.

DeltaSpark, 2025-06-21:
I missed this earlier when I joined the discussion, but it looks like a very similar issue was reported and fixed here: [issue #24141](https://github.com/Koenkk/zigbee2mqtt/issues/24141) . 

A workaround there suggests using an external converter: https://github.com/Koenkk/zigbee2mqtt/issues/24141#issuecomment-2637766446

However, they’ve noted that this fix has regressed for them as well. It might be worth merging these two issues to pool diagnostic information and help find a lasting solution.