Koenkk/zigbee2mqtt #25078: Error „Failed to find an endpoint which supports the OTA cluster“ for 3RTHS0224Z

baerakafubu, 2024-12-05:
### What happened?

Error in log „Failed to find an endpoint which supports the OTA cluster“ for Third Reality [3RTHS0224Z](https://www.zigbee2mqtt.io/devices/3RTHS0224Z.html#third%2520reality-3rths0224z) when searching for OTA

### What did you expect to happen?

After checking for updates the update is starting

### How to reproduce it (minimal and precise)

1. go to OTA in Z2M frontend
2. Check for updates

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

HA add-on on Rpi5-64 4GB

### Debug log

```
Error 2024-12-05 19:15:19z2m: Failed to check if update available for 'Temperatursensor 3RTHS0224Z Wingert' (Failed to find an endpoint which supports the OTA cluster for '0xf442500023ea0000' (3RTHS0224Z))
```

baerakafubu, 2024-12-05:
I also have the problem that this sensor has not reported any change in temperature since the last 4 days. This is the second device within a few weeks with this issue.

Is this related to the error mentioned above or can anyone give me a hint on how to fix this? Are there known issues with similar devices? @Koenkk 
Already tried to re-configure and re-pair but that did not solve the problem.

Koenkk, 2024-12-05:
Does this device support OTA @weihuan1111 ?

azalin1975, 2025-02-10:
I am having similar issues with a couple of my somewhat older Philips Hue v1 dimmers and one of my not so outdated outdoor motion sensor. They either report "Failed to find an endpoint which supports the OTA cluster" or the software claims the dimmer did not respond to the update request. Not sure what, why or how but with de ember driver I have been able to update several devices. So, is there something I can do to fix this or investigate this? Please let me know.
Coördinator type
    EmberZNet

Coordinator revisie
    7.4.4 [GA]

Coordinator IEEE adres
    0xe0798dfffe8f7def

Versie frontend
    0.9.4

zigbee_herdsman_converters_version
    21.31.0

zigbee_herdsman_version
    3.2.5

For what it is worth, it happened with these devices:
- RWL021
- SML002

Fun thing is that the indoor motion sensor was just now updated without any problems.

closed 2025-04-19 by github-actions[bot]