Koenkk/zigbee2mqtt #25080: Watchdog is not activated

acommuni, 2024-12-05:
### What happened?

I have configured the watchdog in configuration.yaml like this

watchdog:
  enabled: true
  interval: 60
  margin: 10

But in the log file there is a message with describing a start without watchdog

_[20:26:35] INFO: Preparing to start...
[20:26:35] INFO: Socat not enabled
[20:26:35] INFO: Starting Zigbee2MQTT...
**Starting Zigbee2MQTT without watchdog.**
[2024-12-05 20:26:41] info: 	z2m: Logging to console, file (filename: log.log)
[2024-12-05 20:26:41] info: 	z2m: Starting Zigbee2MQTT version 1.42.0 (commit #unknown)
[2024-12-05 20:26:41] info: 	z2m: Starting zigbee-herdsman (2.1.9)_

### What did you expect to happen?

I expect that watchdog is activated

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

7.4.4 [GA]

### Adapter

EmberZNet

### Setup

HA on Raspberry 4

Nerivec, 2024-12-05:
Not sure where you got these values from? The configuration of the watchdog also takes place in the add-on configuration, not the yaml.
See https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/blob/master/zigbee2mqtt/DOCS.md#enabling-the-watchdog

acommuni, 2024-12-05:
Could you tell me where is located this file ?

In Add-On page, I have check watchdog too but it seems that it does not to work ("Chien de garde" in french)
[image]

Nerivec, 2024-12-06:
Add-ons > Zigbee2MQTT > Configuration tab > `Show unused optional configuration options` > watchdog field
Follow the instructions from the previous link to configure it.

acommuni, 2024-12-06:
Thank you very much. It works well

closed 2024-12-06 by acommuni

balboafer, 2025-02-14:
Hi! i´ve been trying to configure watchdog following the instructions from this post and i can´t get it to work, can you take a screen shot of the actual watchdog field to see what i´m missing?
Thank you.