Koenkk/zigbee2mqtt #25085: Sonoff ZBMINIR2 lists as unsupported

Jhutjens92, 2024-12-05:
### What happened?

Recently aquired a Sonoff ZBMINIR2. Shows up in Z2M as unsupported however the website lists it as a [supported device](https://www.zigbee2mqtt.io/devices/ZBMINIR2.html). 

`
Zigbee2MQTT:warn  2024-12-05 23:01:49: Device '0x7cc6b6fffef3cd61' with Zigbee model 'ZBMINIR2' and manufacturer name 'SONOFF' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
`

`
Zigbee2MQTT:info  2024-12-05 23:01:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Automatically generated definition","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":2,"category":"config","description":"Ititiate device identification","label":"Identify","name":"identify","property":"identify","type":"enum","values":["identify"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ZBMINIR2","options":[{"access":2,"description":"Sets duration of identification procedure in seconds (i.e., how long device would flash). Value ranges from 1 to 30 seconds (default 3).","label":"Identify timeout","name":"identify_timeout","property":"identify_timeout","type":"numeric","value_max":30,"value_min":1},{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"SONOFF"},"friendly_name":"0x7cc6b6fffef3cd61","ieee_address":"0x7cc6b6fffef3cd61","status":"successful","supported":false},"type":"device_interview"}'
`

`
Zigbee2MQTT:info  2024-12-05 23:01:49: Configuring '0x7cc6b6fffef3cd61'
`

### Zigbee2MQTT version

1.36.0 commit: 86ed71c

### Adapter firmware version

20220928

### Adapter

Slaesh's CC2652RB

### Setup

Docker

crispylees, 2024-12-07:
I've just had a similar issue and resolved it. The steps I took were as follows but I have no idea which (if not all) was the key:

1. Update OS
2. Reboot
3. Wait until all devices are connected and responsive
4. Update zigbee2mqtt
5. Wait until all devices are connected and responsive
6. Delete ZBMINIR2
7. Set ZBMINIR2 back into pairing mode by holding down the only button for 5s until the relay latches and green LED starts flashing
8. Set zigbee2mqtt to permit join
9. Wait until pairing completes and device is recognised as a ZBMINIR2 and supported

Jhutjens92, 2024-12-08:
You can close this issue. Apparently at some point i locked docker at version 1.36. Updated to the latest version and everything works.

closed 2024-12-08 by Jhutjens92

Jens-eh, 2025-01-24:
Hello,
I have same problem. The ZBMINIR2 is considered as unspupported by zigbee2mqtt even though on supported list. 
I do have zigbee2mqtt 1.34.0 and my Raspi apt is not offering any update. I do not use docker. I did reboot.

Still after pairing again I get  "Zigbee model 'ZBMINIR2' and manufacturer name 'SONOFF' is NOT supported"  :-(

The instruction https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html tells me to check Dev console tab of the device and press Generate external definition. But the button is missing.

Any idea to get it working?

Jens

The log: 
info 2025-01-24 18:36:46Device '0x70c59cfffe49447f' joined
info 2025-01-24 18:36:46MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70c59cfffe49447f","ieee_address":"0x70c59cfffe49447f"},"type":"device_joined"}'
info 2025-01-24 18:36:46MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x70c59cfffe49447f"},"type":"device_connected"}'
info 2025-01-24 18:36:46MQTT publish: topic 'zigbee2mqtt/0x70c59cfffe49447f', payload '{"last_seen":"2025-01-24T18:36:44+01:00","linkquality":54}'
info 2025-01-24 18:36:46**Starting interview** of '0x70c59cfffe49447f'
info 2025-01-24 18:36:46MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70c59cfffe49447f","ieee_address":"0x70c59cfffe49447f","status":"started"},"type":"device_interview"}'
info 2025-01-24 18:36:46MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x70c59cfffe49447f"},"type":"pairing"}'
info 2025-01-24 18:36:46MQTT publish: topic 'zigbee2mqtt/0x70c59cfffe49447f', payload '{"last_seen":"2025-01-24T18:36:45+01:00","linkquality":54}'
debug 2025-01-24 18:36:46Device '0x70c59cfffe49447f' announced itself
info 2025-01-24 18:36:46MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70c59cfffe49447f","ieee_address":"0x70c59cfffe49447f"},"type":"device_announce"}'
info 2025-01-24 18:36:46MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x70c59cfffe49447f"},"type":"device_announced"}'
debug 2025-01-24 18:36:46Received Zigbee message from '0x70c59cfffe49447f', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2025-01-24 18:36:46Skipping message, definition is undefined and still interviewing
info 2025-01-24 18:36:46MQTT publish: topic 'zigbee2mqtt/0x70c59cfffe49447f', payload '{"last_seen":"2025-01-24T18:36:45+01:00","linkquality":63}'
info 2025-01-24 18:36:46**Successfully interviewed '0x70c59cfffe49447f', device** has successfully been paired
warning 2025-01-24 18:36:46Device '0x70c59cfffe49447f' with **Zigbee model 'ZBMINIR2' and manufacturer name 'SONOFF' is NOT supported**, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html