Koenkk/zigbee2mqtt #25086: Tuya IH-K009 do not send battery information

Synochy, 2024-12-05:
### What happened?

I paired my Tuya IH-K009 with Zigbee2MQTT and all the information appears but not the battery and voltage information

### What did you expect to happen?

Be able to have the battery % information

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

7.4.5 [GA]

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2 (with last firmware)

### Setup

Add-on HA on orangepi 5B. Supervised version of HA

mainTAP, 2025-01-08:
I see the same behaviour with Tuya IH-K009 ( the rectangular ones powered by 2*AAA )

flogistone, 2025-01-15:
Facing the same behavior... Temp, Humidity, signal strength are ok, but the voltage and battery level..

willmann, 2025-02-09:
same behaviour

adelinchristian, 2025-02-18:
I encounter the same behavior on my side

gyzod, 2025-03-17:
Same thing, 2 out of 3 sensors are not reporting battery readings.

djsakisd, 2025-03-19:
Same problem using H-K009. Below a publish message in mqtt : 
`{
  "battery": null,
  "humidity": 46.76,
  "last_seen": "2025-03-19T17:50:46+02:00",
  "linkquality": 172,
  "temperature": 19.23,
  "voltage": null
}`

BePalm2022, 2025-03-27:
still the same issue , and I see battery is not under device

{
    "last_seen": "2025-03-27T08:53:40+01:00",
    "battery": null,
    "device": {
        "applicationVersion": 70,
        "dateCode": "",
        "friendlyName": "Temperature05_Disponible",
        "hardwareVersion": 1,
        "ieeeAddr": "xxxxxx,
        "manufacturerID": 4417,
        "manufacturerName": "_TZ3000_dowj6gyi",
        "model": "IH-K009",
        "networkAddress": 20640,
        "powerSource": "Battery",
        "stackVersion": 0,
        "type": "EndDevice",
        "zclVersion": 3
    },
    "humidity": 54.1,
    "linkquality": 21,
    "temperature": 22.6,
    "voltage": null
}

dia986, 2025-05-07:
any updates on this?

Synochy, 2025-05-13:
No still waiting

BePalm2022, 2025-05-13:
solved by connecting to the  master ZB dongle again , then place it to the place where it send the data over the slave ZB dongle....not very convenient but worked for me now

Soukyuu, 2025-06-01:
This is definitely a z2m bug. I have switched to ZHA keeping the network (had to confirm that settings have changed) and waited until the sensors sent a new message to show up in ZHA and:

[image]

Soukyuu, 2025-06-01:
So I did the following binding:

[image]

and pressed the reset switch on the sensor within the 10 seconds after clicking on "connect" - the binding was successfully set up and the sensor now shows the battery information:

[image]

Usually it fails because the sensor is in sleep state so the binding request times out. 
In any case, it seems that the binding is required for the sensor to report battery info.

I attempted to make a custom converter and it kind of seems to be working, but it is still complaining about endpoint 19 not being available on re-configuring the device. Still, battery is now shown for those that I reconfigured. Maybe someone with more knowledge can refine this:
```
import {temperature, humidity, battery} from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    fingerprint: [
            ...tuya.fingerprint("TS0201", ["_TZ3000_dowj6gyi", "_TZ3000_8ybe88nf", "_TZ3000_akqdg6g7"]),
            {manufacturerName: "_TZ3000_zl1kmjqx"},
        ],
        model: "IH-K009",
        vendor: "Tuya",
        description: "Temperature & humidity sensor",
        fromZigbee: [fzLocal.TS0201_battery, fz.temperature, fz.humidity],
        toZigbee: [],
        exposes: [e.battery(), e.temperature(), e.humidity(), e.battery_voltage()],
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            // Enables reporting of measurement state changes
            await tuya.configureMagicPacket(device, coordinatorEndpoint);
            await reporting.bind(endpoint, coordinatorEndpoint, [
                "genBasic",
                "genPowerCfg",
                "msTemperatureMeasurement",
                "msRelativeHumidity",
                "manuSpecificTuya_2",
            ]);
        },
        whiteLabel: [tuya.whitelabel("Tuya", "RSH-HS06_1", "Temperature & humidity sensor", ["_TZ3000_zl1kmjqx"])],
};
```

Soukyuu, 2025-10-05:
Unstale

closed 2025-10-05 by Koenkk