Koenkk/zigbee2mqtt #25088: ZY-M100-24GV3 not accurate after the last two upgrades

pvangorp, 2024-12-06:
### What happened?

The devices [ZY-M100-24GV3] (I've got six of them) keep showing presence when nobody is there. A workaround is setting the Move sensitivity and Presence sensitivity to a higher level. E.g. from 0 to 3. The downside is that the presence detection is a lot less accurate. I get away when someone is still there.

### What did you expect to happen?

That when some is presence the devices show present and away when they leave. It worked in the older version

### How to reproduce it (minimal and precise)

Upgrade to 1.42.0-2 and this will occur

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

 7.4.3 [GA]

### Adapter

Sonoff Zigbee 3.0 Dongle Plus-E (EmberZNet)

### Setup

Home Assistant on a MeLE Quieter 3C Fanless Mini PC N5105 (8G/256G)

jesse-kaufman, 2024-12-11:
Same issue here with 1.41.0 onward (including latest-dev)

Also, on this device motion sensitivity and presence sensitivity randomly get reset to 0 and minimum detection distance gets reset to 0.75. I'm guessing that might have something to do with it. It acts like the DPs aren't correct or something.

tam481, 2025-01-17:
I've got the same issue even on the latest (v2.0)

tam481, 2025-03-13:
The last couple of updates have not fixed the issue. It is still not holding the settings. I don't actually think they work at all. Changing the sensitivity and distance doesn't seem to make any difference. They also get reset randomly or whenever my RPi is restarted. It doesn't look like the devs are acknowledging this issue. I'm not sure how to check or raise this!

jesse-kaufman, 2025-03-13:
@tam481 it sounds like it might be an issue with the firmware, itself (https://github.com/Koenkk/zigbee2mqtt/issues/24049#issuecomment-2417846081) so until Tuya fixes that, the issue will persist. I've setup automations in Node-RED to update the settings if it detects that they were reset to their original value. Obviously just a band-aid, but it's kept my Home Assistant logs from filling up with spam about presence and motion sensitivity being outside the range of 1-10

closed 2025-05-21 by github-actions[bot]