Koenkk/zigbee2mqtt #25090: LoraTap SC500ZB opening not correct after some time

skycryer, 2024-12-06:
### What happened?

The Problem i have is everytime i do only partially Close the Cover and directly Open it everything IS fine. But after Like 30 minutes or so it forgets how Long it needs to travel before fully Open again. 

So when i want to Open a partially closed Cover after some time i need to fully Close it first and then Open IT to have it Open the Cover complete again. 

### What did you expect to happen?

Should open correct even without closing it first. 

### How to reproduce it (minimal and precise)

Close a cover partially, leave it like this for 30 minutes or more then open it. It will not fully open again before closing it completely and then open it again.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.3.1.0 build 0

### Adapter

EZSP v12 Sonfoff

### Setup

Addon on Intel NUC

Ny-Mic, 2025-02-06:
I have the same issue under ZIgbee2MQTT 2.1.0

kidekoop2016, 2025-03-11:
Hello. I am looking for shutter module.
1) Is there a solution for the SC500ZB
2) Witch alternative do you recommend?
Thnsk

closed 2025-05-19 by github-actions[bot]