Koenkk/zigbee2mqtt #25091: _TZ3000_typdpbpg not reporting Power correctly

kodifan, 2024-12-06:
### What happened?

Probably after upgrading Z2M to 1.41.0 (fresh install) my Tuya _TZ3000_typdpbpg stared to  report low values for power (for sure, and probably for current as well)
But energy value is populated ok.
Before upgrade everything was reported correctly.  My other plugs (_TZ3000_5f43h46b version) report all values correctly.  
Here is a Home Assistant power graph for my dishwasher. Its not possible dishwasher takes only as much as 42W in peak. Its rather 1500W when it heats water .
Its not Home Assistant issue , Z2m reports low values as well.
[image]
[image]

### What did you expect to happen?

Just have popper power reading reported.

### Zigbee2MQTT version

1.41.0 commit: 69f7282

### Adapter firmware version

20211217

### Adapter

Sonoff CC2652P

### Setup

Plain on RPi 3

kodifan, 2024-12-06:
actually there something very wrong with this, here is picture of this plug is being powered gaming pc right now:
[image]
so i guess it does not report any consumption below some values (high values ) ad then its not correct ...

kodifan, 2024-12-06:
ok i think i narrowed it down, the plug is reporting  more or less correct values but in very long intervals (like 5 minutes or more). 
and that way i cant see high usage when using dishwasher because heating phase is rather quick.
and when i connected pc to this plug, and run bedchamber after a while 180W consumption has been reported.

anyway this behavior is different from previous old (1.22 or something) version of Z2M (it was reported every 5 sec or so if will have to guess) 
any way to fix it if its nit intentional ?

kodifan, 2024-12-16:
@Koenkk can you look at this issue  ?
i checked database.db and only noticeable difference that i can find is acCurrentDivisor attribute:

`{"id":2,"type":"Router","ieeeAddr":"0xa4c138d5d7e19149","nwkAddr":35419,"manufId":4417,"manufName":"_TZ3000_typdpbpg","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0014\u0000\u0000\u0000\u0012 \u0000\u0000\u0000\u0016","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_typdpbpg","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":2000,"acCurrentMultiplier":1,"rmsVoltage":231,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":660}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2690705,"53251":126,"53252":11332,"53253":1337,"powerOnBehavior":1,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024cb032b","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024cb032b","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024cb032b","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":257}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1734344886427}

{"id":3,"type":"Router","ieeeAddr":"0x70b3d52b600fea9d","nwkAddr":17089,"manufId":4660,"manufName":"_TZ3000_5f43h46b","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,6,3,4,5,57345,2820,1794],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_5f43h46b","powerSource":1,"zclVersion":2,"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"500+TZSKT11RM103"}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":233,"rmsCurrent":15,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":361}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"tuyaBacklightMode":1,"moesStartUpOnOff":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":1,"switchType":2}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"500+TZSKT11RM103","zclVersion":2,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1734344886773}
` 

on this "not working" plug its set to 2000, on working one to 1000

kodifan, 2024-12-17:
Something is deeply wrong either with my setup or Z2M in general
I changed reporting max interval to 60 (seconds i guess) and its updating every minute but showing non stop insane high usage of power and current non stop...

[image]

Koenkk, 2024-12-20:
Can you try to stop z2m, in the data/databse.db change the `acCurrentDivisor` start z2m and check if the value is correct now? (the device needs to report the current once for the change to be applied)

kodifan, 2024-12-20:
ok i will try this tomorrow.  but what about those Current readings ? it seems that is globally set to amps instead of miliamps... how to correct this ?

kodifan, 2024-12-21:
> acCurrentDivisor

in dont know how (maybe im not understanding whole process at all) but now this filed is gone form databse.db file

`{"id":2,"type":"Router","ieeeAddr":"0xa4c138d5d7e19149","nwkAddr":35419,"manufId":4417,"manufName":"_TZ3000_typdpbpg","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0014\u0000\u0000\u0000\u0012 \u0000\u0000\u0000\u0016","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_typdpbpg","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentMultiplier":1,"rmsVoltage":233,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":841}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2690705,"53251":126,"53252":11332,"53253":1337,"powerOnBehavior":1,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024cb032b","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024cb032b","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024cb032b","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":360,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":360,"repChange":50},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":360,"repChange":257}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1734772743248}
` 

all ive done is changing max rep interval in reporting tab

Koenkk, 2024-12-21:
> but what about those Current readings 

Change the divisor will change the current readings you see, can you try to force remove the device, re-pair and it and see if the `acCurrentDivisor` comes back?

Nemesis24, 2024-12-22:
Same problem for me it seems reporting mA instead of A, in my case, on one of my 6 devices. 
re-pair never change for the good value

jeremyruffell, 2025-01-15:
I had the same issue, after force removing the device and re-pairing the issue was gone.

closed 2025-03-25 by github-actions[bot]