Koenkk/zigbee2mqtt #25092: Migrated to 2.0.0, failed to start, exited with code 0 or 139

diplix, 2024-12-06:
### What happened?

after i migrated to 2.0.0 (accoring to migration log everything went fine), zigbee2bqtt faily to start:
```
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Starting Zigbee2MQTT without watchdog.
zigbee2mqtt exited with code 0
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Starting Zigbee2MQTT without watchdog.
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Starting Zigbee2MQTT without watchdog.
zigbee2mqtt exited with code 0
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Starting Zigbee2MQTT without watchdog.
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Starting Zigbee2MQTT without watchdog.
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Starting Zigbee2MQTT without watchdog.
zigbee2mqtt exited with code 139
[…]
```

### What did you expect to happen?

error code 139 seems to indicate a serial problem, at least accoring to issue [11545](https://github.com/Koenkk/zigbee2mqtt/issues/11545).

### How to reproduce it (minimal and precise)

i run zigbee2mqtt on docker on raspberry pi 3. 
1.42.0 runs fine.

### Zigbee2MQTT version

1.42.0, 2.0.0

### Adapter firmware version

20240710

### Adapter

zStack3x0, LAUNCHXL-CC1352P-2

### Setup

i run zigbee2mqtt on docker on raspberry pi 3

Koenkk, 2024-12-09:
139 is a SIGSEGV, could you provide your `configuration.yaml`?

diplix, 2024-12-10:
```
homeassistant:
  experimental_event_entities: true
  discovery_topic: homeassistantconf2
frontend: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.1.57:21608
  user: *********
  password: *********
  client_id: zigbee2mqtt_pi
advanced:
  pan_id: ****
  log_directory: data/log
  log_level: error
  log_output:
    - console
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  last_seen: ISO_8601
serial:
  port: /dev/ttyACM0
groups:
  '1':
    friendly_name: kammer_lichter
  '2':
    friendly_name: tint_rgb
    retain: true
    transition: 1
  '3':
    friendly_name: kronleuchter
    retain: true
    transition: 1
  '4':
    friendly_name: kinderzimmer_alles
    retain: true
    transition: 1
  '5':
    friendly_name: kinderzimmer_lichter
    retain: true
    transition: 1
  '6':
    friendly_name: group_tradfri_schalter_2
    retain: true
    transition: 1
  '7':
    friendly_name: schlafzimmer_deckenlampe
    retain: true
    transition: 1
  '8':
    friendly_name: group_lidl_1
  '9':
    friendly_name: group_tradfri_schalter_4
    retain: true
    transition: 1
  '10':
    friendly_name: stube_lichter
    retain: true
    transition: 1
  '11':
    friendly_name: kueche_lichter
    retain: true
[... 385 lines cut ...]
  '0xa4c1380*********':
    friendly_name: bad_decke
    color_sync: false
    transition: 1
  '0x847127f*********':
    friendly_name: tuyai_temp_hum_1
  '0xa4c1383*********':
    friendly_name: Dimmer Bad
    description: Neu, von Tuya
    simulated_brightness:
      delta: 20
      interval: 200
  '0x881a14f*********':
    friendly_name: Tretakt 1
  '0x881a14f*********':
    friendly_name: Tretakt 2
  '0x94b216f*********':
    friendly_name: billy2
blocklist:
  - '0x00124b001d033b3a'
  - '0x0000000000000000'
  - '0x00158d0003ae81d6'
availability: true
version: 2
```

Koenkk, 2024-12-11:
Could you change the log level to debug?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

diplix, 2024-12-12:
not much more info it seems with …
```
advanced:
  log_directory: data/log
  log_level: debug
```

```
pi@olga:~/docker-configs/docker-zigbee2mqtt-v2 $ docker compose logs -f
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Starting Zigbee2MQTT without watchdog.
zigbee2mqtt  | [2024-12-12 05:03:58] info: 	z2m: Logging to console
zigbee2mqtt  | [2024-12-12 05:03:58] debug: 	z2m: Loaded state from file /app/data/state.json
zigbee2mqtt  | [2024-12-12 05:03:58] info: 	z2m: Starting Zigbee2MQTT version 1.42.0-dev (commit #1dde41f7eafe2ecfc55fa1d10a7963c211294043)
zigbee2mqtt  | [2024-12-12 05:03:58] debug: 	z2m: sd-notify loaded
zigbee2mqtt  | [2024-12-12 05:03:58] info: 	z2m: Starting zigbee-herdsman (3.0.2)
zigbee2mqtt  | [2024-12-12 05:03:58] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":xxxx,\"extendedPanID\":[xxx,xxx,xxx,xxx,xxx,xxx,xxx,xxx],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyACM0\"},\"adapter\":{\"disableLED\":false}}"'
zigbee2mqtt  | [2024-12-12 05:03:59] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyACM0","manufacturer":"Texas Instruments","serialNumber":"L1100D19","pnpId":"usb-Texas_Instruments_XDS110__03.00.00.29__Embed_with_CMSIS-DAP_L1100D19-if00","vendorId":"0451","productId":"bef3"},{"path":"/dev/ttyACM1","manufacturer":"Texas Instruments","serialNumber":"L1100D19","pnpId":"usb-Texas_Instruments_XDS110__03.00.00.29__Embed_with_CMSIS-DAP_L1100D19-if03","vendorId":"0451","productId":"bef3"},{"path":"/dev/ttyAMA0"}]
zigbee2mqtt  | [2024-12-12 05:04:00] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"Texas Instruments","serialNumber":"L1100D19","pnpId":"usb-Texas_Instruments_XDS110__03.00.00.29__Embed_with_CMSIS-DAP_L1100D19-if00","vendorId":"0451","productId":"bef3"} => zstack: path=/dev/ttyACM0, score=4
zigbee2mqtt  | [2024-12-12 05:04:00] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":xxxx,"extendedPanID":[xxx,xxx,xxx,xxx,xxx,xxx,xxx,xxx],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json"}'
zigbee2mqtt  | [2024-12-12 05:04:00] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
zigbee2mqtt exited with code 0
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Starting Zigbee2MQTT without watchdog.
zigbee2mqtt  | [2024-12-12 05:03:58] info: 	z2m: Logging to console
zigbee2mqtt  | [2024-12-12 05:03:58] debug: 	z2m: Loaded state from file /app/data/state.json
zigbee2mqtt  | [2024-12-12 05:03:58] info: 	z2m: Starting Zigbee2MQTT version 1.42.0-dev (commit #1dde41f7eafe2ecfc55fa1d10a7963c211294043)
zigbee2mqtt  | [2024-12-12 05:03:58] debug: 	z2m: sd-notify loaded
zigbee2mqtt  | [2024-12-12 05:03:58] info: 	z2m: Starting zigbee-herdsman (3.0.2)
zigbee2mqtt  | [2024-12-12 05:03:58] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":xxxx,\"extendedPanID\":[xxx,xxx,xxx,xxx,xxx,xxx,xxx,xxx],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyACM0\"},\"adapter\":{\"disableLED\":false}}"'
zigbee2mqtt  | [2024-12-12 05:03:59] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyACM0","manufacturer":"Texas Instruments","serialNumber":"L1100D19","pnpId":"usb-Texas_Instruments_XDS110__03.00.00.29__Embed_with_CMSIS-DAP_L1100D19-if00","vendorId":"0451","productId":"bef3"},{"path":"/dev/ttyACM1","manufacturer":"Texas Instruments","serialNumber":"L1100D19","pnpId":"usb-Texas_Instruments_XDS110__03.00.00.29__Embed_with_CMSIS-DAP_L1100D19-if03","vendorId":"0451","productId":"bef3"},{"path":"/dev/ttyAMA0"}]
zigbee2mqtt  | [2024-12-12 05:04:00] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"Texas Instruments","serialNumber":"L1100D19","pnpId":"usb-Texas_Instruments_XDS110__03.00.00.29__Embed_with_CMSIS-DAP_L1100D19-if00","vendorId":"0451","productId":"bef3"} => zstack: path=/dev/ttyACM0, score=4
zigbee2mqtt  | [2024-12-12 05:04:00] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":xxxx,"extendedPanID":[xxx,xxx,xxx,xxx,xxx,xxx,xxx,xxx],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json"}'
zigbee2mqtt  | [2024-12-12 05:04:00] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Starting Zigbee2MQTT without watchdog.
```

diplix, 2024-12-12:
maybe my docker-compose.yaml is of any help — which works flawlessly with 1.42.0

```
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:latest-dev
    restart: unless-stopped
    privileged: true
    volumes:
      - ./data:/app/data
      - /run/udev:/run/udev:ro
    ports:
      # Frontend port
      - 8080:8080
    environment:
      - TZ=Europe/Berlin
    dns:
      - 127.0.0.1 # system resolver
      - 1.1.1.1 # cloudflare
      - 8.8.8.8 # google
    devices:
      - /dev/serial/by-id/usb-Texas_Instruments_XDS110__03.00.00.29__Embed_with_CMSIS-DAP_L1100D19-if00:/dev/ttyACM0
```

djurny, 2024-12-12:
Hi there,
Can you try without the `homeassistant: {}` and `no_occupancy_since: []` parts in your `configuration.yaml` ?
Groetjes,

diplix, 2024-12-12:
deleted lines with `homeassistant: {}` and `no_occupancy_since: []`, but still get the same errors (code 0 or code 139):

```
zigbee2mqtt  | [2024-12-12 11:13:49] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
zigbee2mqtt exited with code 0
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Starting Zigbee2MQTT without watchdog.
[…]
zigbee2mqtt  | [2024-12-12 11:13:49] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Starting Zigbee2MQTT without watchdog.
[…]
zigbee2mqtt  | [2024-12-12 11:14:07] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
zigbee2mqtt exited with code 139
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Starting Zigbee2MQTT without watchdog.
```

diplix, 2024-12-12:
i’m running zigbee2mqtt in a docker container on buster. although 1.42.0 is running smoothly, i’ll try upgrading the pi to bullseye or bookworm in the next couple of days. i guess there might be a possibility that Docker version 20.10.17 doesn’t play nice with the node version that 2.0.0 uses. or is that an unlikely theory?

diplix, 2024-12-13:
just pulled and tried it (an hour after your post), but unfortunately the same error pattern as before :(

Nerivec, 2024-12-14:
@diplix can you try replacing the appropriate "prebuild" file in your container using the ones I built [here](https://github.com/Nerivec/bindings-cpp/actions/runs/12327219169) (make sure to extract and rename the same).
It should be in:
`<zigbee2mqtt folder>`/`node_modules/.pnpm/@serialport+bindings-cpp@12.0.1/node_modules/@serialport/bindings-cpp/prebuilds/xxx/xxx.node`

Then see if Z2M starts?

_Side note: 139 could be a hardware issue too (memory)._

diplix, 2024-12-14:
@Nerivec i would, but can you help/direct me where/how i can find the prebuild files to download?

since zigbee2mqtt version 1.42.0 is running (very) smoothly on the same machine, isn’t it rather unlikly there’s a HW/memory issue?

Nerivec, 2024-12-14:
Scroll down on that page I linked before, it's at the bottom (by arch, like `build-artifacts-linux-arm64`).

Memory is a tricky thing, if it hits an area that wasn't before, and that area is the one with issues... that will do it.
Granted, at this point, it seem serialport is more likely (I figure it's using an improper prebuild).

Koenkk, 2024-12-15:
@Nerivec could you try with the latest dev branch in 30 mins from now?

Nerivec, 2024-12-15:
Wrong tag 😜
@diplix

diplix, 2024-12-15:
i just updated my raspberry pi to bookworm, reinstalled docker, pulled and started my zigbee2mqtt v2.0.0 (latest-dev) container — and now it works. seems to me either v2.0.0 won’t run on older systems / docker installs or memory menagment in buster is not sufficient.

this issue could be closed, but maybe there should be a warning, that v2.0.0 won’t run in older docker installs in the Readme?

Nerivec, 2024-12-15:
latest-dev would have included https://github.com/Koenkk/zigbee2mqtt/pull/25202
Too late to confirm that it works on the old setup though now I guess 😁

diplix, 2024-12-15:
🤷‍♂️

commit 88023e34 (2024-12-13): fix(ignore): Rebuild serialport, attempt to fix #25092
commit 146b6c38 (2024-12-13): fix(ignore): Rebuild serialport, attempt to fix #25092
closed 2024-12-17 by Nerivec