Koenkk/zigbee2mqtt #25093: AVATTO TRV06_1 missing properties exposes

israelcasado, 2024-12-06:
### Link

https://www.zigbee2mqtt.io/devices/TRV06_1.html

### Database entry

{"id":7,"type":"EndDevice","ieeeAddr":"0xa4c138c682be593d","nwkAddr":37281,"manufId":4417,"manufName":"_TZE284_o3x45p96","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE284_o3x45p96","powerSource":3,"zclVersion":3,"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1733487940747}

### Zigbee2MQTT version

1.42.0

### Comments

According to the official documentation this device reports different exposes lock (state), battery_low, etc... but I'm getting only these 
```
{
    "current_heating_setpoint": 14,
    "linkquality": 65,
    "running_state": "idle",
    "system_mode": "heat"
}
```
Do I have to configure anything else?

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_o3x45p96',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I only get these properties
```
{
    "current_heating_setpoint": 14,
    "linkquality": 65,
    "running_state": "idle",
    "system_mode": "heat"
}
```

maunz1337, 2024-12-17:
It seems like this is a new version/type of the TRV.
However i also still dont get all the informations from it.

I bought 10 of these but not all work as designed like the "old ones".

I would really appreciate a fix :)

hayvan96, 2025-02-06:
As for me I don't get battery :
```
{
    "child_lock": "LOCK",
    "current_heating_setpoint": 11.5,
    "frost_protection": "OFF",
    "linkquality": 84,
    "local_temperature": 15,
    "running_state": "idle",
    "scale_protection": "OFF",
    "system_mode": "heat"
}
```

falkgeist, 2025-03-28:
I seem to have the same problem, since the beginning, but with random exposes. The first time I added one of the devices, everything was there, after I reconfigured and interviewed it (with the buttons in the Web UI). Then I added the second one, had multiple issues which I resolved by reinstalling Z2M, afterwards the devices were not only missing the battery but also the local_temperature:
```
{
    "child_lock": "LOCK",
    "current_heating_setpoint": 22,
    "frost_protection": "ON",
    "linkquality": 21,
    "scale_protection": "ON",
    "system_mode": "auto"
}
```
It is usable like this, but not really convenient. It would be majorly useful to know the actual temperature.

lu-mueller, 2025-04-16:
I don't get battery low.

```
{
    "child_lock": "LOCK",
    "current_heating_setpoint": 8,
    "frost_protection": "OFF",
    "last_seen": "2025-04-16T14:31:02+02:00",
    "linkquality": 204,
    "local_temperature": 20,
    "local_temperature_calibration": 0,
    "pi_heating_demand": 0,
    "running_state": "idle",
    "scale_protection": "OFF",
    "system_mode": "heat"
}

```

simi-1984, 2025-04-19:
Same Problem.

Battery low Status is missing.

Martin123456777, 2025-05-27:
Same problem.
Battery low Status is missing.

falkgeist, 2025-07-27:
actually, yes, I'm also missing the battery_low and it's quite annoying to have the thermostats lose power without any notice...

leszegfu, 2026-01-15:
I don't see the battery status either. 

BTW Did anyone manage to get the schedule to work on 'auto' mode on TS0601 _TZE284_o3x45p96 device? I believe it was working fine via Tuya app when I initially set it up, but after I hooked it up in Z2M it doesn't seem to take into account the schedule in any way. I've set up an automation to set the temperatures manually (using heat mode), but it would be great to use the built-in schedule in case I need to tweak it in the future.

{
    "child_lock": "UNLOCK",
    "current_heating_setpoint": 22,
    "frost_protection": "ON",
    "linkquality": 204,
    "local_temperature": 22,
    "pi_heating_demand": 0,
    "running_state": "idle",
    "scale_protection": "ON",
    "schedule_friday": "06:00/25.0 09:00/25.0 10:00/25.0 11:00/25.0 12:00/25.0 18:00/25.0",
    "schedule_monday": "06:00/25.0 09:00/25.0 10:00/25.0 11:00/25.0 12:00/25.0 18:00/25.0",
    "schedule_saturday": "06:00/25.0 09:00/25.0 10:00/25.0 11:00/25.0 12:00/25.0 18:00/25.0",
    "schedule_sunday": "06:00/25.0 09:00/25.0 10:00/25.0 11:00/25.0 12:00/25.0 18:00/25.0",
    "schedule_thursday": "06:00/25.0 09:00/25.0 10:00/25.0 11:00/25.0 12:00/25.0 18:00/25.0",
    "schedule_tuesday": "06:00/25.0 09:00/25.0 10:00/25.0 11:00/25.0 12:00/25.0 18:00/25.0",
    "schedule_wednesday": "06:00/25.0 09:00/25.0 10:00/25.0 11:00/25.0 12:00/25.0 18:00/25.0",
    "system_mode": "heat"
}

Z2M version: 2.7.2
HA version: 2026.1.1

filiphavlin, 2026-02-03:
Also same issue

MichaelHainbach, 2026-02-26:
Jonstep related my issue #31156 to this one. Perhaps my finding helps?
In short I have two hardware version of the Avatto TRV6 - a version 1 and a version 1b.
On one the Battery works - but the Thermostate valve shows null - on the other the battery shows null, but the valve metric works.... perhaps that helps on how to interpret??

falkgeist, 2026-09-05:
If I ever find the time to understand, apply and test the proposed fix, I will report the results, but I can't promise anything in regards to when I'll get around to do it.

MichaelHainbach, 2026-09-09:
...was hoping one would try that knows what he/shes doing... anyhow. I implemented the External converter - and the Avatto now shows TRV06_1_battery_fix as model - the TRV6_1b remained as was.
I removed batteries for a while and the the device re-joined - but did not send battery information - DP102 (battery_low) and DP35 (fault) are absent from that wake-up dump.
I...battery remained unknown.
@Jonstep101010 - claude suggests to try to connect to a tuya bridge and confirm the values - would that help - I do not have one right now - but would be ready to buy one... I also want to get the valve states from the TRV6_1b version..

MichaelHainbach, 2026-10-11:
For everyone missing `battery_low` on the TRV06_1 (`_TZE284_o3x45p96`): I debugged this on 3x TRV06_1 and 6x TRV06_1b and wrote it up in #33350.

What I found on the TRV06_1:
- The valve does **not report DP35 (the `fault` bitmap, bit 0 = low battery) or the weekly programs on its own**. A setpoint or mode change only produces DP2/3/4/101, never DP35.
- In zigbee-herdsman-converters 26.115.1 (shipped with Z2M 2.14.2) DP35 is mapped as `[35, null, tuya.valueConverter.errorOrBatteryLow]`, i.e. without a property name, and the generic fromZigbee ignores the boolean that converter returns (`Object.assign`). Only `error` is published.

What worked for me (external converter): name the datapoint `battery_low`, test bit 0, and ask the valve for all datapoints periodically:

```js
[35, "battery_low", {from: (v, meta, options, publish) => { publish({error: v}); return (v & 1) === 1; }}]
// ...
extend: [tuya.modernExtend.tuyaBase({dp: true, timeStart: "2000", queryOnConfigure: true, queryOnDeviceAnnounce: true, queryIntervalSeconds: 6 * 3600})],
```

After that all three TRV06_1 received their first `battery_low` value exactly one query interval (6 h) after the restart. I introduced the naming fix and the query together, so I can't say how much each contributed. Not tested: battery impact of the 6 h query.

Details, the Tuya datapoint table for the TRV06_1b and debug excerpts are in #33350.