Koenkk/zigbee2mqtt #25096: [Wrong device]: MiBoxer FUT035Z+ shown as TuYa TS0502B

lverti, 2024-12-06:
### Link

https://www.zigbee2mqtt.io/devices/FUT035Z_.html

### Model

FUT035Z+

### Description

Dual white LED controller

### Vendor

MiBoxer

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/FUT035Z+.png

### Database entry

{"id":30,"type":"Router","ieeeAddr":"0x8c65a3fffe5650d1","nwkAddr":14022,"manufId":4098,"manufName":"_TZB210_ue01a0s2","powerSource":"Mains (single phase)","modelId":"TS0502B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,4096,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":0,"65534":0,"modelId":"TS0502B","manufacturerName":"_TZB210_ue01a0s2","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":71,"powerSource":1}},"lightingColorCtrl":{"attributes":{"57344":1000,"57600":3874126036992,"61453":1,"colorCapabilities":16,"tuyaRgbMode":0}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"61440":902}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":71,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1733499617660}

dawiinci, 2024-12-21:
Same here. I thought this was fixed but it is not. 

https://github.com/Koenkk/zigbee2mqtt/issues/16768

dawiinci, 2024-12-25:
I see a lot of issues latrly with devices recognized as Tuya products. Is there a correlation?

Kapteenivamma, 2024-12-27:
Same problem here. They worked very well for 6months and they were recognized correctly as FUT035Z+.

In some point all of the 3pcs of FUT035Z+ lost abilty for light transition and also some times they only light up with minimum brightness with motion detector.

I started to investigate and find out that they are recognized as TS0502B.

dawiinci, 2024-12-27:
Same for the FUT037Z+.

Kapteenivamma, 2025-01-05:
Updating to Zigbee2MQTT 2.0.0 did not fix this problem

closed 2025-04-14 by github-actions[bot]