Koenkk/zigbee2mqtt #25101: Some devices turn on after restarting container

abrasher, 2024-12-06:
### What happened?

When I restart the container, some lights that were previously off prior to the restart suddenly turn on. 

This was not happening on version 1.4.1

I also get an error about publishEntityState

### What did you expect to happen?

The devices to not turn on after restarting

### How to reproduce it (minimal and precise)

Turn light off in
docker compose restart zigbee2mqtt
Light turns on 

### Zigbee2MQTT version

1.42.0-dev commit: 639e3bb6fd50928da002d417455b5eaf3acb2ad8

### Adapter firmware version

20220219

### Adapter

zStack3x0 Sonoff Zigbee USB Stick

### Setup

Docker, linux x64, debian 12

### Debug log

The devices that turn on after restart are:
- 0x001788010bbe88be (kitchen lamp, hue bulb)
- 0x0017880103d7e478 (fan kitchen, different hue bulb)
- 0xa4c13860c237ce1d (bedroom_closet, gledopto led controller)

[log.log](https://github.com/user-attachments/files/18042061/log.log)

abrasher, 2024-12-07:
I guess this is intended behaviour as the official docs have it ignore if the previous state is unavailable

```
conditions:
          - condition: template
            value_template: "{{trigger.from_state.state != 'unavailable'}}"
          - condition: template
            value_template: "{{trigger.to_state.attributes.event_type == 'single'}}"
```

closed 2024-12-07 by abrasher

mundschenk-at, 2024-12-10:
The actual state is always a timestamp (generated by HA). I agree that having to check for `unavailable` is somewhat unfortunate, but that's how the HA state engine seems to work (unless we introduce a new `idle` `event_type`).