Koenkk/zigbee2mqtt #25103: 1.42.0 is very laggy

joschaschultze, 2024-12-06:
### What happened?

When I switch from version 1.41.0 to 1.42.0 my whole zigbee network is getting really laggy and unresponsive. I've tried it several times: 1.41.0 is very fast together with domoticz but as soon as I load 1.42.0 every device is very laggy.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20240710

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Setup

Docker

redmss, 2024-12-07:
The same thing happens to me, since this version something has become slower, even some devices communicate late, I notice it quite a lot because I have many zigbee pir detectors with zigbee lighting, before it was immediate now it sometimes takes up to two seconds to turn on on some occasions

WendelFelius, 2024-12-07:
Me too. On a Pi 5, Sonoff ZBDongle-P, Z2M 1.42.0-2. Latest software everywhere. Lots of devices showing different states, don't respond, some show offline. It's a mess.

phil-schneider, 2024-12-09:
I have the same feeling.
My innr lights react like two seconds Kater.

mackowskim, 2024-12-09:
Same here, also timeouts in z2m logs and pairing issues

mill9r, 2024-12-10:
It's my first experience with rpi5 and Zigbee. Could you please highlight, which version was stable? 
I've installed 1.42.0 and from time to time have a stack of errors.

IceColdCrash, 2024-12-17:
Same here, some devices work perfectly, others take a very long time to respond

closed 2025-02-23 by github-actions[bot]