Koenkk/zigbee2mqtt #25113: Migration to 2.0 doesn't work with read-only configuration.yml

rhuss, 2024-12-07:
### What happened?

When using configuration.yaml in a read-only mode (e.g. when mounting the file from a Kubernetes secret), then I get the following error with the latest `dev` builds:

```
11:24 ➜ k logs zigbee2mqtt-0
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
Migration notes written in /app/data/migration-1.x.x-to-2.0.0.log
Error: EROFS: read-only file system, open '/app/data/configuration.yaml'
    at Object.openSync (node:fs:596:3)
    at Object.writeFileSync (node:fs:2322:35)
    at Object.writeIfChanged (/app/lib/util/yaml.ts:43:12)
    at write (/app/lib/util/settings.ts:208:10)
    at Object.apply (/app/lib/util/settings.ts:447:5)
    at Object.migrateIfNecessary (/app/lib/util/settingsMigration.ts:482:18)
    at start (/app/index.js:136:27)
```

AFAIU, a possible migration, replaces `configuration.yaml` in-place, which doesn't work in this setup. Would it be possible to allow an _offline_ migration path, where one could migrate on the configuration file from the CLI and switch off live migration?

This would also be helpful for GitOps scenarios where the source of truth is a Git repository (in-place live updates don't fit in this paradigm, either).

### What did you expect to happen?

I expect that `configuration.yaml` is not written to. I think one of the reason why multiple `device` entries are allowed for exactly to support this scenario, like in :

```
devices:
    - devices-run.yaml
    - devices.yaml
```

so that new devices get added to the first (writable) file.

### How to reproduce it (minimal and precise)

Make `configuration.yml` read-only and start zigbee2mqtt.

### Zigbee2MQTT version

dev-latest

### Adapter firmware version

20230507

### Adapter

Slaesh CC2652/CC1352

### Setup

K3s cluster on RasPi 4

rhuss, 2024-12-07:
Somewhat related: https://github.com/Koenkk/zigbee2mqtt/issues/2970

Koenkk, 2024-12-09:
I'm not sure how to properly solve this, z2m assumes the config is writeable (also when updating the config via the webui). I guess the least we can do is not crash z2m when the write fails (CC: @jlpedrosa)

rhuss, 2024-12-10:
For my use case, it would be good enough to have an 'offline migration' mode, e.g., to run z2m on the CLI (not as daemon) to convert a given file to the new format that I could then use as a replacement for the secret. This would be a good workaround, but I agree that, in general, one should only allow humans and machines to write into the same file. Clear ownership on the file level helps to avoid conflicts where easily. In z2m 1.0, that works nicely for me (after #2970 was resolved)

Alternatively, I could also imagine to have an option like 

```yaml
migrationFile: /path/to/a/writable/file
```

that would write the given file in case of a migration that then later could be picked up and put into a `Secret`. (I'm using k3s on raspi's and z2m as a `StatefulSet`)

rhuss, 2024-12-10:
ah, I see. It's not only about migration but also for regular operation.

@Koenkk, what's about a solution for writing newly detected devices? I.e., allow an external file to be included. That file could be written, but people could also take over parts of it and commit those later to a version control system. That would also be a good compromise. I like the `devices:` approach (again, as leveraged in #2970, where new devices are added to the first file on the list, and not in configuration.yaml directly)

Koenkk, 2024-12-11:
@rhuss it's already possible to write devices to a separate file (like `devices.yaml`), see https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#extract-config-to-separate-files

rhuss, 2024-12-12:
> @rhuss it's already possible to write devices to a separate file (like `devices.yaml`), see https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#extract-config-to-separate-files

Yes, I know, and I actively use [this](https://github.com/Koenkk/zigbee2mqtt/issues/2970#issuecomment-734912951) since it is available. This allowed me to move the well-known devices in a read-only `devices.yaml` (stored as a `Secret` in my k3s cluster and Git for version control) while still being able to auto-discover new devices and storing it in that one writable `devices-run.yaml`. My setup looks like 

```yaml
devices:
  # File for new detected devices:
  - devices-run.yaml
  # File mounted read-only from a K8s secret
  - devices.yaml
```

That approach works very nicely for devices. My point for this issue is that the same or a similar technique is helpful for other configuration values that can be added or changed during runtime, like user-provided values from a GUI. As for devices, the whole pattern allows two files to be merged during runtime. I would take that as a blueprint.

For example, one could use a global include for the whole configuration file, like

```yaml
include: configuration-ui.yaml
...
```

In this case, any change during runtime would be written to `configuration-ui.yaml`. Values in this file take precedence over the values in this file.

In some sense, this resembles the way how a [UnionFS](https://en.wikipedia.org/wiki/UnionFS) works that are prominently used, e.g., for Docker container image layers: In a Docker (OCI) container, the whole image is mounted as read-only layers and one read-write layer on the top. All layers are merged so the user can have a unified view on the filesystem. The same could be done similarly for `configuration.yaml`, with two layers only.

vxav, 2025-01-08:
I just had the same error, it was fine `<2.0.0` though.
I was thinking of moving the file to PVC with an init container but that would negate any changes made with the UI afaik.
Did you end up with a working setup here?

rhuss, 2025-01-09:
> Did you end up with a working setup here?

Nope, I still stick to ZigBee 1.x as long as I can't find a workable solution for my current setup.

vxav, 2025-03-16:
I ended up doing something similar as well. I'd rather maintain UI usability out of continuity but I also don't change settings there so it'll do.

closed 2025-05-23 by github-actions[bot]