Koenkk/zigbee2mqtt #25114: Aqara T1M CL-L02D Group does not control endpoints

Mitchy99, 2024-12-07:
### What happened?

After the 1.42.0 update the Z2M group toggle doesn''t change the included Aqara T1M device.
Both endpoint 1 an 2 do not respond
When the devices are switched directly the coupled Z2M group is showing the correct response.

### What did you expect to happen?

Toggle or On/Off action to the Z2M group which includes the Aqara T1M / CL-L02D should stitch the devices correctly

### How to reproduce it (minimal and precise)

Downgrade to 1.41.0 does solve the issue

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

0x26780700

### Adapter

conbee II

### Setup

HA OS

### Debug log

[2024-12-07 09:56:05] error: 	z2m: No converter available for 'state' ("OFF")

grant17, 2024-12-08:
I am having the same issue with a tuya dimmer "TS0052", also downgraded and now is working again

mundschenk-at, 2024-12-08:
Probably a uplicate of #24792.

closed 2025-02-14 by github-actions[bot]