Koenkk/zigbee2mqtt #25116: Zigbee Network Instability and Connectivity Issues

mbenecke, 2024-12-07:
### What happened?

Random Zigbee devices lose connection to Home Assistant:

-     Door contact sensors do not report open/close events.
-     TRVs fail to receive target temperature updates according to the schedule.
-     Device interviews fail for previously functioning devices (e.g., well-known sensors are not detected anymore).

This instability undermines the reliability of the Zigbee network, even though the setup appears optimal.

### What did you expect to happen?

-     Consistent and reliable communication between Zigbee devices and Home Assistant.
-     Door sensors to report states promptly.
-     TRVs to follow schedules without issue.
-     Device interviews to succeed as expected.

### How to reproduce it (minimal and precise)

1.     Use a Zigbee network with a CC2652P-based coordinator (SLZB-06), Zigbee2MQTT in an LXC container on Proxmox, and multiple routers spread throughout a reinforced concrete house.
2.     Run Zigbee on channel 11, alongside a Philips Hue mesh on channel 25. Wi-Fi is configured to use channels 3-11.
3.     Introduce additional Zigbee devices (e.g., holiday lights connected via plugs).
4.     Monitor connectivity and device behavior over time.

### Zigbee2MQTT version

1.42.0 (commit: 5b907f2f)

### Adapter firmware version

20240710

### Adapter

SLZB-06 (CC2652P)

### Setup

    Zigbee2MQTT running as a containerized LXC on a Proxmox cluster (2x Intel N97).     Coordinator centrally located in the house and connected via PoE LAN to a gigabit switch.     Building features reinforced concrete walls and roofs, requiring routers in    Zigbee2MQTT running as a containerized LXC on a Proxmox cluster (2x Intel N97).     Coordinator centrally located in the house and connected via PoE LAN to a gigabit switch.     Building features reinforced concrete walls and roofs, requiring routers in every room.     86 Zigbee devices (52 end devices, 34 routers). every room.     86 Zigbee devices (52 end devices, 34 routers).

Ricc68, 2024-12-07:
Check wifi-zigbee channels for coexistence and avoid overlaps.

mbenecke, 2024-12-07:
Thanks for the suggestion, acording metageek [ZigBee and Wi-Fi Coexistence](https://www.metageek.com/training/resources/zigbee-wifi-coexistence/)  my Zigbee2mqtt on Zigbee-channel 11, alongside a Philips Hue mesh on Zigbeechannel 25. Wi-Fi is configured to use Wi-Fichannels 3-11.

Ricc68, 2024-12-07:
Channel 25 zb is overlapping with channel 11 WiFi sidelobe. Channel 11 zg should be ok. But have you checked neighbours WiFi channels? You can do so with the Android WiFi analyser app.

mbenecke, 2024-12-08:
Thank you for the feedback!

### Wi-Fi Analysis:
I used the UniFi WiFi Analyzer app and verified that my neighbors' Wi-Fi networks are mostly on channel 1, with very low signal strength due to the reinforced concrete (see the attached screenshot). My APs are configured between channels 3 and 11. I chose Zigbee channel 25 for the Philips Hue mesh, assuming minimal overlap, but I understand there could still be interference from the Wi-Fi sidelobe. However, Philips Hue runs rock-solid with 50 devices, most of them mains-powered. This are my Neighbors:
[image]

### Antenna Placement:
The Zigbee coordinator’s antenna is positioned centrally and is at least 1 meter away from any Wi-Fi devices, USB 3.0 devices, or other sources of potential interference.

### Serial Coordinator Over LAN:
I wasn’t aware this could be a potential issue. I’ll review the debug logs for anomalies related to the serial-over-LAN setup and see if anything points to instability. Would you suggest any specific debug settings or parameters to monitor this more effectively?

Let me know if you have further suggestions or if I should prioritize testing any particular change first. Thanks again for the input!

Ricc68, 2024-12-08:
I see WiFi channel 1 is crowded, it doesn't count much that transmissions are weak because also ZigBee transmissions maybe weak. It overlaps with ZigBee channel 11.
If it was me, I would opt for channel 25 for ZigBee and channel 26 for hue if it supports it (channel 26 is not always supported).
Weakness of ZigBee channel 11 may very well be the cause of random disconnections. This would be my first experiment: try moving the network on channel 25 and 26.

Yes, the serial is a time-bound protocol and ethernet is the opposite as it allows for significant timing jitter. If there are problems probably in the log you may find timeouts or sync errors. Enable the debug in z2m and observe what happens when a disconnect happens.

phil-schneider, 2024-12-11:
Thanks a lot for your answer and support
I dont know what I did. After reinstalling an old backup and again a newer backup...
Now it everything seems to work from my side.

Ricc68, 2024-12-12:
@mbenecke how's going with your problem? Have you solved it?

mbenecke, 2024-12-12:
Thanks, @Ricc68 . I haven’t had the time to implement your recommendation yet. With Christmas around the corner, you know how it goes. I’ll keep you posted!

closed 2025-02-18 by github-actions[bot]