Koenkk/zigbee2mqtt #25118: [New device support]: TS0601 by _TZE200_eqpaxqdv KnockautX cover controller

nabossha, 2024-12-07:
### Link

https://www.brelag.com/shop/pims3028-knockautx-plugin-empfanger-multi-stak-3-stas-3-4517

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0xa4c13845fc80e081","nwkAddr":23801,"manufId":4417,"manufName":"_TZE200_eqpaxqdv","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u000b\u000b�.g","65506":54,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_eqpaxqdv","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1733580674889}

### Zigbee2MQTT version

1.42.0

### Comments

i could get the cover to work on the commands up/down/stop with both of the following device-definitions - in both cases the position-indicator of the cover was not reflecting the actual state, it was going from 0-100 or the other way around in a matter of seconds where the real change of the cover-position was maybe around 1-2% of the distance.
according to the KnockautX-manual the position should be controllable via the KnockautX-app (which I do not have since using the app requires an additional KnockautX-gateway...)

**TS0601_cover_8** 
- controlling up/down/stop worked fine, though the position of the cover was off probably to missing settings of the bottom/upper limit

**TS0601_cover_4**
- controlling  up/down/stop worked fine, though the position of the cover was off probably to missing settings of the endpoints, was not able to make bottom/upper limit work (did not have any influence of the displayed cover-position)

### External definition

```shell
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{ modelID: 'TS0601', manufacturerName: '_TZE200_eqpaxqdv' },],
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_eqpaxqdv',
    description: 'Roller Blinds Motor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.enum('motor_direction', ea.STATE_SET, ['normal', 'reversed']).withDescription('Set the motor direction'),
        e.numeric('motor_speed', ea.STATE_SET).withValueMin(0).withValueMax(255).withDescription('Motor speed').withUnit('rpm'),
        e.enum('opening_mode', ea.STATE_SET, ['tilt', 'lift']).withDescription('Opening mode'),
        e.enum('set_upper_limit', ea.SET, ['SET']).withDescription('Set the upper limit, to reset limits use factory_reset'),
        e.enum('set_bottom_limit', ea.SET, ['SET']).withDescription('Set the bottom limit, to reset limits use factory_reset'),
        e.binary('factory_reset', ea.SET, true, false).withDescription('Factory reset the device'),
    ],
    whiteLabel: [
        tuya.whitelabel('Tuya', 'TS0601', 'Cover', ['_TZE200_eqpaxqdv'])
    ],
    configure: async (device, coordinatorEndpoint) => {
        device.powerSource = 'Mains (single phase)';
        device.save();
    },
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({OPEN: tuya.enum(0), STOP: tuya.enum(1), CLOSE: tuya.enum(2)})],
            [2, 'position', tuya.valueConverter.coverPosition],
            [3, 'position', tuya.valueConverter.raw],
            [5, 'motor_direction', tuya.valueConverter.tubularMotorDirection],
            [7, null, null], // work_state, not useful, ignore
            [101, 'opening_mode', tuya.valueConverterBasic.lookup({tilt: tuya.enum(0), lift: tuya.enum(1)})],
            [102, 'factory_reset', tuya.valueConverter.raw],
            [103, 'set_upper_limit', tuya.valueConverter.setLimit],
            [104, 'set_bottom_limit', tuya.valueConverter.setLimit],
            [105, 'motor_speed', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

//     tuya.whitelabel('Tuya', 'TS0601', 'Cover', ['_TZE200_eqpaxqdv']),
```

### What does/doesn't work with the external definition?

**works:**
- up / down / stop

**does not work**
- setting upper / bottom limits
- setting cover the a %-position

I tried using some additional information about the device-clusters available here but could not get anything else to work, lacking experience of debugging zigbee-stuff
https://github.com/zigpy/zha-device-handlers/issues/3463

hubaksis, 2025-01-13:
Hi,
Did you resolve the issue with setting the up/bottom limits?

I've got TS0601_cover_8 (ZM25TQ-2/25, mains powered motor) in addition to my battery powered ones.

I've upgraded Zigbee2Mqtt to v2.0.0, and the TS0601_cover_8 is recognized. However, I am not able to set limits - my existing remotes (Zemismart YR2118B) do not connect with this motor.

I would appreciate any help.

hubaksis, 2025-01-14:
Sorry, my previous comment was for a different device, I found this issue only because my motor was also displayed as `TS0601_cover_8`.
While this might be not a solution for your motor, I found a command which worked for my motor to set limits. Could it be the same for yours? The issue was that values for `103` must be boolean.

add to exposes 
```
  e.enum('upper_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Set or Reset the upper stroke limit').withCategory('config'),
  e.enum('middle_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Set or Reset the middle stroke limit').withCategory('config'),
  e.enum('lower_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Set or Reset the lower stroke limit').withCategory('config'),
  e.enum('motor_working_mode', ea.STATE_SET, ['continuous', 'intermittently']).withDescription('Motor operating mode').withCategory('config'),
```

and to meta
```
  [103, 'upper_stroke_limit', tuya.valueConverterBasic.lookup({SET: true, RESET: false})],
  [104, 'middle_stroke_limit', tuya.valueConverterBasic.lookup({SET: true, RESET: false})],
  [105, 'lower_stroke_limit', tuya.valueConverterBasic.lookup({SET: true, RESET: false})],
  [106, 'motor_working_mode', tuya.valueConverterBasic.lookup({continuous: tuya.enum(0), intermittently: tuya.enum(1)})]

```

nabossha, 2025-01-14:
@hubaksis thanks for your update. Meanwhile I have contacted the vendor of the device asking for support regarding the datapoints but they were not very helpful (or in my opinion do not even know what I was asking for)
I will test your settings in the next days and update this thread again!

closed 2025-03-24 by github-actions[bot]