Koenkk/zigbee2mqtt #25124: TS011F, _TZ3000_w0qqde0g - sw_version:500+TZSKT21Z3102

garry0garry, 2024-12-08:
### What happened?

TS011F, _TZ3000_w0qqde0g
Zigbee2MQTT version 1.42.0
Coordinator type EZSP v13
Coordinator revision 7.4.4.0

Version are not read correctly:
`"sw_version":"500+TZSKT21Z3102"`

### What did you expect to happen?

A valid version number was expected.

### How to reproduce it (minimal and precise)

Always reproduces.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.4.4.0

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus V2” (model "ZBDongle-E"), EZSP v13

### Setup

Docker

### Debug log

https://github.com/Koenkk/zigbee2mqtt/issues/22182#issuecomment-2494751649

garry0garry, 2024-12-14:
What logs can I collect to make corrections to the code?

alexsotoaguilera, 2024-12-14:
[2024-12-14 13:27:08] info: 	z2m: Checking if update available for 'Smart plug - Office'
[2024-12-14 13:27:08] debug: 	zhc:ota:common: Checking if an update is available for '0x70b3d52b60109201' (TS011F)
[2024-12-14 13:27:08] debug: 	zhc:ota:common: Using endpoint '1'
[2024-12-14 13:27:08] debug: 	zh:controller:endpoint: CommandResponse 0x70b3d52b60109201/1 genOta.imageNotify({"payloadType":0,"queryJitter":100}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"})
[2024-12-14 13:27:08] debug: 	zh:ember: ~~~> [ZCL to=0x70b3d52b60109201:12641 apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":234,"commandIdentifier":0}]
[2024-12-14 13:27:08] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=72 Len=26]
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=0 frmRx=2](ackRx=0)
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=0 frmTx=1)
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=2](frmRx=2) Added to rxQueue
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3](ackRx=1)
[2024-12-14 13:27:08] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=72 Len=7]
[2024-12-14 13:27:08] debug: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=24 messageTag=34 status=OK]
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=3](frmRx=3) Added to rxQueue
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=1)
[2024-12-14 13:27:08] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=72 Len=22]
[2024-12-14 13:27:08] debug: 	zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=DIRECT], [indexOrDestination=12641], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":24}], [messageTag=34]
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=4](frmRx=4) Added to rxQueue
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5](ackRx=1)
[2024-12-14 13:27:08] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=89:"INCOMING_ROUTE_RECORD_HANDLER" Seq=72 Len=18]
[2024-12-14 13:27:08] debug: 	zh:ember:ezsp: ezspIncomingRouteRecordHandler(): callback called with: [source=12641], [sourceEui=0x70b3d52b60109201], [lastHopLqi=196], [lastHopRssi=-51], [relayCount=0], [relayList=]
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=5](frmRx=5) Added to rxQueue
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6](ackRx=1)
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=6](frmRx=6) Added to rxQueue
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7](ackRx=1)
[2024-12-14 13:27:08] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=98:"undefined" Seq=72 Len=13]
[2024-12-14 13:27:08] debug: 	zh:ember:ezsp: <=x= Ignored unused/unknown [CBFRAME: ID=98:"undefined" Seq=72 Len=13]
[2024-12-14 13:27:08] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=72 Len=30]
[2024-12-14 13:27:08] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":3392,"groupId":0,"sequence":211}], [packetInfo:{"senderShortId":12641,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":200,"lastHopRssi":-50,"lastHopTimestamp":0}], [messageContents=10ea0b00c3]
[2024-12-14 13:27:08] debug: 	zh:controller: Received payload: clusterID=25, address=12641, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=200, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":234,"commandIdentifier":11},"payload":{"cmdId":0,"statusCode":195},"command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}}
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-12-14 13:27:08] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[... cut ...] ":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":53,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":16,"attrData":1}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-12-14 13:27:25] debug: 	z2m: Received Zigbee message from 'Smart plug - Bedroom', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0

garry0garry, 2024-12-14:
Neither your nor my logs contain any information about the firmware version. Am I wrong?

garry0garry, 2024-12-21:
Zigbee2MQTT version 1.42.0
Coordinator type EmberZNet
Coordinator revision 7.4.5 [GA]

Same result for _TZ3000_w0qqde0g:
Firmware version 500+TZSKT21Z3102

garry0garry, 2025-01-04:
Zigbee2MQTT version 2.0.0
Coordinator type EmberZNet
Coordinator revision 7.4.5 [GA]

_TZ3000_w0qqde0g
Firmware version 500+TZSKT21Z3102

garry0garry, 2025-02-08:
Zigbee2MQTT version 2.1.1
Coordinator type EmberZNet
Coordinator revision 7.4.5 [GA]

_TZ3000_w0qqde0g
https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html
Firmware version 500+TZSKT21Z3102

closed 2025-04-18 by github-actions[bot]