Koenkk/zigbee2mqtt #25125: No converter available for 'state' ("ON")

elronzo, 2024-12-08:
### What happened?

After updating from 1.41.0 to 1.42.0 I do get the following error when trying to switch an Aqara CL-L02D on or off:

No converter available for 'state' ("ON")

The device has a state attribute with values ON and OFF. It can be controlled over HomeAssistant because it sends a different payload than FHEM (sending only state on or off).

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

Containerized Z2M

elronzo, 2024-12-08:
Might be related to Issue  #25114

grant17, 2024-12-08:
hello i am having the same issue with a tuya dimmer switch "TS0052" 

If I control the dimmer directly it works fine but I have a few of them in a zigbee group and they no longer work in the group, when trying to control the group from home assistant i get the error z2m: No converter available for 'state' ("ON")"

petermnt, 2024-12-08:
Running into the same problem in a group with 2x https://www.zigbee2mqtt.io/devices/046677585235.html#philips-046677585235
One "special" thing about this device is that it has 2 light entities on different endpoints (11 and 12).
I'm also having problems trying to edit the group.  Removing endpoint 11 worked fine but I can't remove the ones on endpoint 12.
After deleting the group I even got weirder errors.

My other groups/devices seem fine.

mundschenk-at, 2024-12-08:
These reports are likely duplicates of #24792.

Koenkk, 2024-12-10:
Likely a duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/24792 indeed.

closed 2024-12-10 by Koenkk

elronzo, 2025-01-11:
I resolved the issue by sending "state_white" instead of "state".