Koenkk/zigbee2mqtt #25126: Mumubiz TYZGTH1CH-D1RF

mercur1al, 2024-12-08:
### What happened?

Auto mode does not work

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

20230507

### Adapter

zStack3x0 СС2652P

### Setup

add-on

### Debug log

z2m: No converter available for 'manual_mode' ("OFF")
z2m: No converter available for 'auto_settings' ({"enabled":"on","temp_greater_then":"off","temp_greater_value":43.5,"temp_lower_then":"on","temp_lower_value":29.2})

mercur1al, 2024-12-16:
up

rejoe2, 2024-12-20:
Same Problem here .
Zigbee2MQTT-Version
    [1.42.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.42.0) commit: [861cba6](https://github.com/Koenkk/zigbee2mqtt/commit/861cba6)

Coordinator-Typ
    zStack3x0

Coordinator-Version
    20240710

closed 2025-02-26 by github-actions[bot]