Koenkk/zigbee2mqtt #25131: Aqara door sensore MCCGQ11LM disconnect

developerGM, 2024-12-08:
### What happened?

My Aqara MCCGQ11LM door sensor disconnects Always from zignee network.
I have tried to connect again ma every time After about 2 hours is disconnected.
i changed battery, position but nothing. I dont know what can I do.

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

20230507

### Adapter

Sonoff dongle plus

### Setup

Home Assistant in Raspberry pi4

lux73, 2024-12-09:
i can't second this - all of my Aqara Door Sensors works a expected

CC2652RB - FW 20240710 - z2m 1.42.0 on bare Metal - Debian 12.7

lux73, 2024-12-09:
please keep in mind that Transmit Power only describes the direction from the Coordinator to the Devices

if the Devices are **not powerful to answer** they will fall off your Network

i could recommend to build a strong Zigbee Mesh with many Routers around

developerGM, 2024-12-09:
So I need to set to 20 the Trasmit power, restart Zigbee2MQTT and pair again the device?

developerGM, 2024-12-10:
I ma not solved, yesterday after a few hourse the device was disconnected againg from network.
I don't know why...I think that this problem starts when I have changed router wifi because before I don't never have this disconnections.
Is strange that only this device is disconnected, all others 65 devices are all perfectly online

developerGM, 2024-12-10:
My channel is 11
and my wifi is 2.4Ghz and 5Ghz, but this is the same settings that I have with old router too.

closed 2025-02-17 by github-actions[bot]