Koenkk/zigbee2mqtt #25132: zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO

cikaybe, 2024-12-08:
### What happened?

Hi,

I’m having this error. I’m using a slzb-06 dongle. How can I stop it?

mgmtPermitJoinRsp after 10000ms
[2024-12-08 13:45:08] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 15:15:08] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 16:05:08] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 16:41:48] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 17:18:28] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 18:38:28] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 18:48:28] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 19:58:28] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms

### What did you expect to happen?

The error to stop. 

### How to reproduce it (minimal and precise)

It’s a fresh installation with a slzb-06 dongle. 

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

20240315

### Adapter

SLZB-06

### Setup

Raspberry 4 8GB with Home Assistant

### Debug log

[2024-12-08 07:55:08] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 08:28:28] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 09:15:08] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 11:15:08] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 11:38:45] error: 	z2m: Publish 'set' 'state' to 'L Z2 Corredor Baixo (2)' failed: 'Error: ZCL command 0xa4c1383be8ffde7e/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 54832 - 1 - 49 - 6 - 11 after 10000ms)'
[2024-12-08 11:55:08] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 11:58:28] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 13:45:08] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 15:15:08] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 16:05:08] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 16:41:48] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 17:18:28] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 18:38:28] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 18:48:28] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 19:58:28] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-08 21:31:24] error: 	z2m: Publish 'set' 'state' to 'L Z2 Corredor Cima (2)' failed: 'Error: ZCL command 0xa4c1386962c430da/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 19287 - 1 - 36 - 6 - 11 after 10000ms)'
[2024-12-08 22:31:48] error:

vlavrynovych, 2024-12-11:
Hi guys, I upgraded from 1.35.1-1 to 1.42.0-2

I have a similar problem, but for me **Permit All** button **does not work** at all. The request sends successfully from frontend and no reaction - no counter. After playing with it I've found that the `sensor.zigbee2mqtt_bridge_permit_join_timeout` is `Unknown`.

And I also see similar error in the logs about Timeout after 10000ms with similar payload, but in my case it belongs to lumi.ctrl_neutral2 model. This error happens ocasionally and is not related to the attempt to permit joining

vlavrynovych, 2024-12-11:
BTW, if I try to restart the addon then joining is allowed, but I cannot turn it off. 
Until I update lets say `select.zigbee2mqtt_bridge_log_level` to another value it's stuck in the selected position and now errors and no reaction.

Not sure if it's related or maybe just another symptome but I also cannot build the Map

vlavrynovych, 2024-12-11:
I see that `permit_join_timeout` will be removed in https://github.com/Koenkk/zigbee2mqtt/pull/25082 (already see it in the [changelog](https://gist.github.com/Koenkk/bfd4c3d1725a2cccacc11d6ba51008ba)). I hope it will solve my problem(s).

Nerivec, 2024-12-12:
@cikaybe if you have `permit_join: true` in `configuration.yaml`, try removing the line (will default to `false`). _That line will be gone in 2.0.0 anyway._ Looks like you are having other issues though, since it seems everything is timing out. Did you go through the [checklist](https://www.zigbee2mqtt.io/advanced/zigbee/02_improve_network_range_and_stability.html) for network stability?

@vlavrynovych that timeout is likely just the device being unavailable from time to time (passing interference?). Try to remove the `permit_join` line from your `configuration.yaml` (same as above).

vlavrynovych, 2024-12-12:
Hi @Nerivec, 

My problem is fixed.

I removed `permit_join` and applied this config (from [Zigbee2MQTT 2.0.0 breaking changes](https://github.com/Koenkk/zigbee2mqtt/discussions/24198))
```yaml
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false
```

Now even network map works!! So, means it was somehow related. 
Thank you @Nerivec , you saved me a lot of time.

cikaybe, 2024-12-12:
@Nerivec thanks for taking the time for answering.

About my configuration.yalm:
```
homeassistant: true
advanced:
  network_key:
    - lots of numbers
  pan_id: 18035
  ext_pan_id:
    - lots of numbers
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: error
mqtt:
  base_topic: zigbee2mqtt2
  server: mqtt://core-mosquitto
  user: *
  password: *
serial:
  port: tcp://192.168.68.126:6638
  adapter: zstack
  baudrate: 115200
frontend:
  port: 8099
device_options:
  legacy: false
devices:
  -devices
permit_join: false
availability: true
groups:
  '1':
    friendly_name: L Z Corredor Cima
```

I have a fresh intallation with a SLZB-06 with lan on a raspberry 4 8gb. 

Today logs:
```
[2024-12-11 00:41:49] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 00:48:29] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 02:15:09] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 02:25:09] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 02:55:09] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 04:05:09] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 04:51:49] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 04:55:09] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 07:18:29] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 08:11:49] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 08:42:28] error: 	zh:controller:device: Read response to 0xa4c1383d242e8418 failed (ZCL command 0xa4c1383d242e8418/1 genTime.readRsp({"localTime":787221701}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":150,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240)))
[2024-12-11 08:58:29] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 09:45:09] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 10:38:29] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 11:15:09] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 11:18:29] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 12:05:09] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 12:08:29] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 12:41:49] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 12:56:20] error: 	z2m: Publish 'set' 'state' to '0xa4c1382c179b1755' failed: 'Error: ZCL command 0xa4c1382c179b1755/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
[2024-12-11 12:56:58] error: 	z2m: Publish 'set' 'state' to '0xa4c1382c179b1755' failed: 'Error: ZCL command 0xa4c1382c179b1755/2 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
[2024-12-11 13:20:49] error: 	zh:controller: Interview failed for '0xa4c1383b808177ec with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms'
[2024-12-11 13:20:49] error: 	z2m: Failed to interview '0xa4c1383b808177ec', device has not successfully been paired
[2024-12-11 13:20:58] error: 	zh:controller: Interview failed for '0xa4c1383b808177ec with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms'
[2024-12-11 13:20:58] error: 	z2m: Failed to interview '0xa4c1383b808177ec', device has not successfully been paired
[2024-12-11 13:23:02] error: 	zh:controller: Interview failed for '0xa4c1383b808177ec with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms'
[2024-12-11 13:23:02] error: 	z2m: Failed to interview '0xa4c1383b808177ec', device has not successfully been paired
[2024-12-11 13:25:09] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-11 13:38:29] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[... 10 lines cut ...]
[2024-12-11 21:28:29] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[... cut ...]
[2024-12-12 14:45:09] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-12 14:55:09] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-12 15:05:09] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
[2024-12-12 15:21:49] error: 	zh:controller: Failed to keep permit join alive: Error: AREQ - ZDO - mgmtPermitJoinRsp after 10000ms
```

cikaybe, 2024-12-12:
I'm not adding anything not manually or through an automation.

The code I get after restarting. Should I wait a couple of hours?
[45df7312_zigbee2mqtt_2024-12-12T20-39-44.220Z.log](https://github.com/user-attachments/files/18117350/45df7312_zigbee2mqtt_2024-12-12T20-39-44.220Z.log)

The restart part starts at line 6195.

Koenkk, 2024-12-13:
@cikaybe could you provide the debug log from **starting** z2m until this error appears? The log is truncated and therefore I cannot see the root cause.