Koenkk/zigbee2mqtt #25133: z2m: EventBus error 'HomeAssistant/publishEntityState': Cannot read properties of null (reading 'toString')

XenorPLxx, 2024-12-08:
### What happened?

Hi,

I'm running in docker and my about shows as 1.42.0-dev commit: [7d63a87aad954225c77156d22576939b1ffe7d6c](https://github.com/Koenkk/zigbee2mqtt/commit/7d63a87aad954225c77156d22576939b1ffe7d6c)

I have experimental HA events turned on.

I've went trough breaking changes list and I'm pretty sure I addressed all issues, and everything seems to be behaving correctly within HA, but I am seeing errors in Z2M log like the one below. Not sure if that's something I should worry about, but I see that this code was in fact changed recently here https://github.com/Koenkk/zigbee2mqtt/commit/61cf5094c7f0bf112374b900a13a3a5986519004.

```
[2024-12-09 00:12:21] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Switch_Hallway-Lights', payload '{"action":null,"device_temperature":27,"flip_indicator_light":"ON","led_disabled_night":false,"linkquality":145,"mode_switch":"anti_flicker_mode","operation_mode":"control_relay","power_outage_count":2,"power_outage_memory":true,"state":"OFF"}'
[2024-12-09 00:12:21] error:    z2m: EventBus error 'HomeAssistant/publishEntityState': Cannot read properties of null (reading 'toString')
[2024-12-09 00:12:21] debug:    z2m: TypeError: Cannot read properties of null (reading 'toString')
    at HomeAssistant.onPublishEntityState (/app/lib/extension/homeassistant.ts:1284:50)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:23)
    at EventEmitter.emit (node:events:529:35)
    at EventBus.emitPublishEntityState (/app/lib/eventBus.ts:151:22)
    at Controller.publishEntityState (/app/lib/controller.ts:328:23)
    at publish (/app/lib/extension/receive.ts:155:17)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:181:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17)
```

### How to reproduce it (minimal and precise)

I'm not sure how to reproduce, it happens 'on its own' without any triggers from my side, in my setup with z2m docker connected to HA.

### Zigbee2MQTT version

1.42.0-dev commit: 7d63a87aad954225c77156d22576939b1ffe7d6c

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee USB CC2652P (zStack3x0)

### Setup

plain docker on unraid, connected to HA

### Debug log

[log.log](https://github.com/user-attachments/files/18054400/log.log)

ccutrer, 2024-12-10:
I'm also seeing this as of commit https://github.com/Koenkk/zigbee2mqtt/commit/6905c10a05479bef762138b18b4d6fd14e992c25. Disabling Event entities didn't seem to help. MQTT wasn't getting any actual state updates, so I've reverted to 1.42.0 for now.

Koenkk, 2024-12-10:
Could you try to stop z2m, remove the `action` from the `data/state.json` for that device, start z2m and see if the issue is fixed?

Belox86, 2024-12-10:
i have the same problem (or very similar).
i'm 
running  1.42.0-dev commit: [acba533](https://github.com/Koenkk/zigbee2mqtt/commit/acba533)
with experimental_event_entities true

my debug log is a bit different:
```
[2024-12-05 21:18:28] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Multi Switch Serra', payload '{"action":null,"battery":91,"linkquality":204,"voltage":2900}'
[2024-12-05 21:18:28] error: 	z2m: EventBus error 'HomeAssistant/publishEntityState': Cannot read properties of null (reading 'toString')
[2024-12-05 21:18:28] debug: 	z2m: TypeError: Cannot read properties of null (reading 'toString')
    at HomeAssistant.onPublishEntityState (/app/lib/extension/homeassistant.ts:1287:50)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:23)
    at EventEmitter.emit (node:events:531:35)
    at EventBus.emitPublishEntityState (/app/lib/eventBus.ts:151:22)
    at Controller.publishEntityState (/app/lib/controller.ts:326:23)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at runNextTicks (node:internal/process/task_queues:64:3)
    at processImmediate (node:internal/timers:449:9)
    at publish (/app/lib/extension/receive.ts:155:17)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:181:13)
```
i look in the state.json (in addon_configs\45df7312_zigbee2mqtt_edge\zigbee2mqtt ) but i didn't find any action.
[state.json](https://github.com/user-attachments/files/18086265/state.json)

XenorPLxx, 2024-12-10:
yup, no action in mine either
```
root@Home-unRaid:/mnt/user/appdata/zigbee2mqtt# cat state.json | grep action
root@Home-unRaid:/mnt/user/appdata/zigbee2mqtt#
```
```json
root@Home-unRaid:/mnt/user/appdata/zigbee2mqtt# cat state.json
{
    "0x00158d00029bbade": {
        "angle": -14.32,
        "battery": 90,
        "linkquality": 225,
        "side": 0,
        "voltage": 2985,
        "temperature": 23,
        "current": 0,
        "power": 26,
        "device_temperature": 24,
        "power_outage_count": 334
    },
    "0x00158d0002f386cc": {
        "battery": 97,
        "linkquality": 123,
        "voltage": 2995,
        "temperature": 31,
        "device_temperature": 28,
        "power_outage_count": 21
    },
    "0x00158d0002f38750": {
        "battery": 100,
        "linkquality": 178,
        "voltage": 3015,
        "temperature": 29,
        "device_temperature": 28,
        "power_outage_count": 181
    },
    "0x00158d00042853c3": {
        "consumption": 17.550312042236328,
        "energy": 17.55,
        "linkquality": 222,
        "power": 42.7,
        "state": "OFF",
        "state_l1": "OFF",
        "state_l2": "ON",
        "temperature": 35,
        "voltage": 225.45,
        "update_available": false,
        "update": {
            "state": "idle",
            "installed_version": 47,
            "latest_version": 47
        },
        "power_outage_count": 664,
        "current": 0.19,
        "device_temperature": 35,
        "interlock": false
    },
    "0x00158d00044939f7": {
        "battery_low": false,
        "gas": false,
        "gas_density": 0,
        "tamper": false,
        "sensitivity": "high",
        "linkquality": 102
    },
    "0x00158d00049fb9c6": {
        "battery": 77,
        "humidity": 53.42,
        "linkquality": 127,
        "temperature": 22.48,
[... 569 lines cut ...]
    "0x54ef44100080f4f8": {
        "device_temperature": 23,
        "power_outage_count": 2,
        "state": "OFF",
        "voltage": 230,
        "current": 0,
        "power": 0,
        "consumer_connected": true,
        "linkquality": 138,
        "energy": 0.87,
        "consumption": 0.8717289566993713
    },
    "0x28dba7fffe0171e4": {
        "battery": 100,
        "voltage": 3000,
        "update": {
            "state": "idle",
            "installed_version": 8704,
            "latest_version": 8704
        },
        "update_available": false,
        "linkquality": 120
    }
}
```

Koenkk, 2024-12-11:
Fixed in https://github.com/Koenkk/zigbee2mqtt/pull/25169!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. Note that the dev branch currently contains **breaking changes**, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

closed 2024-12-11 by Koenkk