Koenkk/zigbee2mqtt #25154: A lot of devices pairing with unkown vendor

devbysb, 2024-12-10:
### What happened?

Hi I try to add devices to zigbee2mqtt with a slzb-06m coordinator but when I open pairing mode. I have a lot of device near 150 added with unkown vendor.
[image]

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

latest 

### Adapter

slzb-06

### Setup

docker compose

Chartreusito, 2024-12-10:
Hi, i guess you're french as suggested by your screenshot.

At 150+ devices, i would advise you to split your network in half with another instance of zigbee2mqtt running on a second coordinator (on a different channel).

That's what i did a few weeks ago and it solved loads of different problems for me.

I would advise you to go for a recommended network adapter as SLZB06 if you can.

vlavrynovych, 2024-12-11:
Hi @Chartreusito , I think that @devbysb is talking not about his real 150+ devices but about somehow randomly generated by mistake

devbysb, 2024-12-12:
Hi @vlavrynovych  you are totally right. I have reinstall z2m create new network and block all random device and pairing all. It's ok all is working. But I don't understand why I have a lot of random devices...

closed 2025-02-18 by github-actions[bot]