Koenkk/zigbee2mqtt #25157: Ikea SYMFONISK volume hold release

remidebette, 2024-12-10:
### What happened?

Hi, 
I seem to understand that the IKEA [SYMFONISK sound remote gen2](https://www.zigbee2mqtt.io/devices/E2123.html) events has the following event_types: `volume_up_hold`, `volume_down_hold`

In my automation, for hold buttons, I usually do a repeat loop with an until based on a hold release event.

### What did you expect to happen?

Other IKEA controllers like the [RODRET](https://www.zigbee2mqtt.io/devices/E2201.html) or the [STYRBAR](https://www.zigbee2mqtt.io/devices/E2001_E2002.html) have hold event_types but also a hold release (like `brightness_stop`, `arrow_left_release`, `arrow_right_release`)

Is it something that does not exist in the hardware? Or could it be added in z2m?

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.4.4

### Adapter

EmberZNet

### Setup

Activated events

remidebette, 2025-02-10:
Not stale

IKIKN, 2025-03-04:
I reproduce the same issue

closed 2025-05-11 by github-actions[bot]

remidebette, 2025-06-04:
Not stale