Koenkk/zigbee2mqtt #25164: SLZB-06M cant pair devices anymore

panhans, 2024-12-11:
### What happened?

My network is very stable and now I tried to pair a new device with my coordinator. But it doesn't switch to pairing mode. The adapter is bound over LAN.

### What did you expect to happen?

Enabling pairing mode without issues.

### How to reproduce it (minimal and precise)

Select the coordinator an click on *permit join*.

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

 7.4.5 [GA]

### Adapter

SLZB-06M 

### Setup

Home Assistant Addon on x86 Machine

### Debug log

`z2m: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Delivery failed for '5101'.'`

Nerivec, 2024-12-14:
Have you identified that device 5101 (0x13ed)? Is the error always with that device?

panhans, 2024-12-14:
For some reason z2m detected my coordinator as a dongle-e. I resetup and repaired all devices and now everything is fine. Maybe this causes the issue.

closed 2024-12-14 by panhans