Koenkk/zigbee2mqtt #25170: Joining a certain new device instantly crashes zig2mqtt - every time.

HenrikBork, 2024-12-11:
### What happened?

Joining new device "Lightbee" "Zigbee Smart Switch" from "LightSolutions.  (A 4 button battery operated device, with a very useful physical design - compatible with standard power grid switches)
zigbee2mqtt instantly crashes, every time.

Contex: zig2mqtt running in this setup for a year (rather stable). Coordinator for ~25 zigbee devices. Roughly 1-10 messages per second. (average <1/s). Just updated zig2mqtt today - for other device, I do not think that is related to the issue.

### What did you expect to happen?

I expected it to join or maybe be rejected - but zig2mqtt should stay alive.

### How to reproduce it (minimal and precise)

press any button on "Lightbee" device

### Zigbee2MQTT version

1.42.0 commit: bc6eccf

### Adapter firmware version

6.10.3.0 build 297

### Adapter

1a86:55d4 QinHeng Electronics SONOFF Zigbee 3.0 USB Dongle Plus V2

### Setup

on Pi running: Raspbian GNU/Linux trixie/sid

### Debug log

[2024-12-11 20:01:08] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/PowerSw6', payload '{"current":0.08,"energy":0,"linkquality":164,"power":7.5,"state":"ON","voltage":233}'
[2024-12-11 20:01:08] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/PowerSw8', payload '{"current":0,"energy":0,"linkquality":168,"power":0,"state":"ON","voltage":234}'
[2024-12-11 20:01:08] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/PowerSw7', payload '{"current":0,"energy":0,"linkquality":112,"power":0,"state":"ON","voltage":235}'
[2024-12-11 20:01:08] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/PowerSw2', payload '{"linkquality":172,"state":"ON","update":{"installed_version":52442738,"latest_version":53163636,"state":"available"}}'

RangeError: The value of "value" is out of range. It must be >= 0 and <= 65535. Received 24221335
    at checkInt (node:internal/buffer:74:11)
    at writeU_Int16LE (node:internal/buffer:724:3)
    at Buffer.writeUInt16LE (node:internal/buffer:732:10)
    at BuffaloZdo.writeUInt16 (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:40:21)
    at BuffaloZdo.buildIeeeAddressRequest (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/zspec/zdo/buffaloZdo.ts:906:14)
    at Function.buildRequest (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/zspec/zdo/buffaloZdo.ts:767:32)
    at Controller.identifyUnknownDevice (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:542:40)
    at Controller.onZclPayload (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:850:37)
root@xxxxxpi:/opt/zigbee2mqtt# 

Explanation: first 4 lines where "stuff working fine", I waited a few seconds before pressing a button. The error message happens instantly.

HenrikBork, 2024-12-11:
Wireshark files using NordicSemi 802.15.4 sniffer.

wireshark_go_1.pcapng : captured while z2m was dead before sniffing
wireshark_go_3_crash_z2m.pcapng : captured while z2m was running (and crashed)
[dumps.zip](https://github.com/user-attachments/files/18102559/dumps.zip)

HenrikBork, 2024-12-12:
Any pointers as where to start looking for a solution would be appreciated :-) (I am new to this code)

Koenkk, 2024-12-13:
You are using the old `ezsp` driver, it's recommended to use the new `ember` drivers ([info](https://github.com/Koenkk/zigbee2mqtt/discussions/21462)), could you check if the issue still happens with that version?

HenrikBork, 2024-12-15:
Updating the dongle to "ncp-uart-hw-v7.4.5.0-zbdonglee-115200.gbl" and changing to ember resolved the issue.
Thanks a lot. :-)

closed 2024-12-15 by HenrikBork