Koenkk/zigbee2mqtt #25175: BHT-006 thermostat reporting multiple times per second

jvdburgt, 2024-12-12:
### What happened?

I have 2 Moes (Tuya) BHT-006 thermostats (floor heating type). I just discovered that they both flood the network with about 4-5 messages per second, multiples with the same payload). This causes network issues and also results in the devices becoming unresponsive after a while.

There was a similar problem [reported ](https://github.com/Koenkk/zigbee2mqtt/issues/19045) with certain Tuya based presence sensors, which has since been resolved with a FW update.

I'm hoping the same can be done here.

### What did you expect to happen?

The device only reports when attributes (significantly) change.

### How to reproduce it (minimal and precise)

Add them to HA via Zigbee2Mqtt and look at the logs.

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

7.3.2.0

### Adapter

SkyConnect

### Setup

Add-on on HA on x86-64

lux73, 2024-12-13:
your z2m Version is from 12/2021...

please update to latest & recheck the behavior

chernyshevruslan, 2024-12-18:
Did you find a solution to the problem? I have a similar problem with the thermostat  [TZE204_tagezcph](https://github.com/Koenkk/zigbee2mqtt/issues/25234).

MrdUkk, 2024-12-19:
@chernyshevruslan i think you can grab firmware replacement here and flash it  https://habr.com/ru/articles/864740/

toat, 2025-02-22:
> [@chernyshevruslan](https://github.com/chernyshevruslan) i think you can grab firmware replacement here and flash it https://habr.com/ru/articles/864740/

Hi, 

I got 3 BHT-006, all flood my network and crash them regulary. 

I flashed firmware with link above, seems to work, no more flood.

jvdburgt, 2025-02-22:
How did you flash it?

toat, 2025-02-22:
> How did you flash it?

https://github.com/slacky1965/tuya_thermostat_zrd

I follow step by step. At the end of success firmware update, i repair the module to have the correct name and firmware version (restart z2m) between delete (force) and pair again.

closed 2025-05-01 by github-actions[bot]