Koenkk/zigbee2mqtt #25180: Startup error logs on Tuya TS0726

anyhow-sg-smarthome, 2024-12-13:
### What happened?

Device: https://www.zigbee2mqtt.io/devices/TS0726.html#tuya-ts0726

Getting the following error logs during startup:
```
z2m: Failed to configure 'Switch', attempt 2 (TypeError: Cannot read properties of undefined (reading 'bind')
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:24)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:6504:33)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
```

database.db
```
{"id":3,"type":"Router","ieeeAddr":"0xa4c138698192b54e","nwkAddr":34797,"manufId":4417,"manufName":"_TZ3002_1s0vfmtv","powerSource":"Mains (single phase)","modelId":"TS0726","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":4,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0,"switchMode":0}},"genBasic":{"attributes":{"65503":"�?�.\u0012","65506":54,"65508":0,"65534":0,"appVersion":82,"manufacturerName":"_TZ3002_1s0vfmtv","zclVersion":3,"modelId":"TS0726","powerSource":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x847127fffedfe258","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":4,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchMode":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x847127fffedfe258","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1734051945326}

```

### What did you expect to happen?

Error logs should not show during startup

### How to reproduce it (minimal and precise)

Start Zigbee2Mqtt

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

7.4.4 [GA]

### Adapter

SLZB-07 ember

### Setup

Add-on on HA green

Koenkk, 2024-12-13:
Could you provide the data/database.db entry of your device?

anyhow-sg-smarthome, 2024-12-14:
Hi @Koenkk, my initial post has the db entry. Did I post the wrong one? I have gotten it from z2m data folder

Koenkk, 2024-12-15:
This is a 4 gang switch: https://www.zigbee2mqtt.io/devices/TS0726.html#tuya-ts0726, does your version also have 4 gangs or is it only 2 ?

anyhow-sg-smarthome, 2024-12-15:
yeah I am quite surprised that z2m is seeing my 2-gang & 3-gang switches as 4-gang which have errors. I am unsure if it is whether it is unsupported in z2m (but it works perfectly currently for me) or the switch manufacturer decided to place 4-gang for all kind of switches. If these error logs are something I need to live with till I change them, I will close the issue. Thanks for clarifying!

Koenkk, 2024-12-17:
I added the device now!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. Note that the dev branch currently contains **breaking changes**, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `TS0726_2_gang.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2024-12-17 by Koenkk