Koenkk/zigbee2mqtt #25193: Color_mode wrongly reported by some light

chris-1243, 2024-12-13:
### What happened?

Since last dev (1.42.0-dev commit: [1dde41f7eafe2ecfc55fa1d10a7963c211294043](https://github.com/Koenkk/zigbee2mqtt/commit/1dde41f7eafe2ecfc55fa1d10a7963c211294043)), it seems the color_mode is wrongly reported in HA. The color_mode is always "xy". This would be fine for an RGB light but not for a color temperature capable light only like [INNR RB 249 T](https://www.zigbee2mqtt.io/devices/RB_249_T.html#innr-rb%2520249%2520t).

### What did you expect to happen?

Light which is not RGB capable not to set `color_mode` "xy" but only "color_temp"

### Zigbee2MQTT version

1.42.0-dev commit: 1dde41f7eafe2ecfc55fa1d10a7963c211294043

### Adapter firmware version

20240710

### Adapter

Sonoff Dongle-P

### Setup

Z2M on Docker

### Debug log

[log.txt](https://github.com/user-attachments/files/18132626/log.txt)
[home-assistant.log](https://github.com/user-attachments/files/18132627/home-assistant.log)

Koenkk, 2024-12-15:
I initially understood the issue wrong, could you provide the debug log when it goes to `color_mode: xy`?

chris-1243, 2024-12-15:
You should have some date in my last debug log. It happens right after the pairing. 

The ieee address of the device is `0x9c69370000cbca15`, Innr E14 dimmable + color temp

alexsotoaguilera, 2024-12-16:
I hace the same issue with other device: https://github.com/Koenkk/zigbee2mqtt/issues/24963

Koenkk, 2024-12-17:
> It happens right after the pairing.

Does it only happen right after pairing, or does it also happen after setting the color temp once? If yes, could you provide the log of that?

chris-1243, 2024-12-18:
After the pairing. When you set a `color_temp` value, the error is not present anymore

alexsotoaguilera, 2024-12-18:
I changed color_temp and the device stops to report XY.

Thanks chris-1243

Koenkk, 2024-12-21:
Found and fixed the issue!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. Note that the dev branch currently contains **breaking changes**, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

closed 2024-12-21 by Koenkk

coradia, 2025-01-16:
I'm able to reproduce this issue with [Tuya TS0502B](https://www.zigbee2mqtt.io/devices/TS0502B.html#tuya-ts0502b) in the form of [LUUMR Smart LED GX53 lights](https://www.lights.co.uk/p/luumr-smart-led-gx53-9w-cct-zigbee-hue-set-of-3-10018159.html).

Zigbee2MQTT version: [2.0.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0) commit: [060ae99](https://github.com/Koenkk/zigbee2mqtt/commit/060ae99)
Coordinator type: EmberZNet
Coordinator revision: 8.0.2 [GA]
Coordinator IEEE Address: 0x9035eafffeb85676
Frontend version: 0.9.4
zigbee-herdsman-converters version: 21.12.0
zigbee-herdsman version: 3.2.1

[home-assistant_mqtt_2025-01-16T02-53-52.941Z.log](https://github.com/user-attachments/files/18432792/home-assistant_mqtt_2025-01-16T02-53-52.941Z.log)
[z2mqtt.log](https://github.com/user-attachments/files/18432810/z2mqtt.log)