Koenkk/zigbee2mqtt #25199: [New device support]: Tuya TRV _TZE204_ogx8u5z6

Soulfly999, 2024-12-14:
### Link

https://www.banggood.com/Tuya-Smart-Zigbee3_0-Thermostat-Radiator-Valve-Temperature-Schedule-Setting-Window-Open-Detection-Function-APP-Remote-Voice-Control-Work-with-Alexa-Google-Assistance-p-2020881.html?rmmds=myorder&cur_warehouse=CN

### Database entry

{"id":50,"type":"EndDevice","ieeeAddr":"0xa4c1388e0d126e59","nwkAddr":9570,"manufId":4417,"manufName":"_TZE204_ogx8u5z6","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"�\u0000\u0000\u0000i\u0006\u0001\u0000\u0000i\u000f\u0001\u0000\u0000i\u0018\u0001\u0000\u0000i!\u0001\u0000\u0000i+\u0001\u0000\u0000i5\u0001\u0000\u0000i?\u0001\u0000\u0000i","65506":56,"65508":0,"appVersion":74,"modelId":"TS0601","manufacturerName":"_TZE204_ogx8u5z6","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1734186010073}

### Zigbee2MQTT version

1.42.0

### Comments

I need ZM support for the this Tuya TRV please.

### External definition

```shell
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_ogx8u5z6',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Cant see something!?
[image]

Soulfly999, 2024-12-14:
Looks like it's the same [Tuya TRV TS0601](https://www.zigbee2mqtt.io/devices/TS0601_thermostat_1.html)

redbull-80, 2024-12-14:
Hallo, ich habe die selben Thermostate und wünsche mir auch die volle Unterstützung.
Vielen dank im voraus!

Hi, I have the same thermostats and would love to get full support too. Thanks a lot!

ApunktK, 2024-12-14:
teilt das ding, je mehr Leute es benötigen, desto schneller bekommen wir es.

wlfbck, 2024-12-14:
Feature-wise it seems identical to all the other TS0601, atleast i don't see a difference between what's available [here](https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html) and in the manual that came with it.

Soulfly999, 2024-12-17:
Is it possible to implementiert these and howong would it take ca? Otherwise I have to Look after other TRVs and sell these

Appolon20, 2024-12-17:
same issue here,
[image]
i´ve no idea how to implement this into z2m in Home Assistant

Bommel24, 2024-12-18:
I have the same issue with these Thermostats I can test and provide answers I just need someone to guide me :D

ApunktK, 2024-12-18:
> Ist es möglich, diese zu setzen und wie würde es ca? Ansonsten muss ich nach anderen TRVs Ausschau halten und diese verkaufen

https://amzn.eu/d/dgRFlIb nimm die gehen direkt

cbshopper, 2024-12-18:
this is my solution - just copied from "TS0601_thermostat_3" out of tuja.js:

```const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
        zigbeeModel: ['_TZE204_ogx8u5z6'], // Hier die Hersteller-ID
    
        fingerprint: tuya.fingerprint('TS0601', [
           '_TZE204_ogx8u5z6'
        ]),
        model: 'TS0601_thermostat_3',
        vendor: 'Tuya',
        description: 'Thermostatic radiator valve',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
    
        onEvent: tuya.onEventSetTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.child_lock(),
            e.battery_low(),
            e
                .climate()
                .withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
                .withSystemMode(['auto', 'heat', 'off'], ea.STATE_SET)
                .withRunningState(['idle', 'heat'], ea.STATE)
                .withLocalTemperatureCalibration(-9, 9, 1, ea.STATE_SET),
            ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
            e
                .binary('scale_protection', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('If the heat sink is not fully opened within ' +
                'two weeks or is not used for a long time, the valve will be blocked due to silting up and the heat sink will not be ' +
                'able to be used. To ensure normal use of the heat sink, the controller will automatically open the valve fully every ' +
                'two weeks. It will run for 30 seconds per time with the screen displaying "Ad", then return to its normal working state ' +
                'again.'),
            e
                .binary('frost_protection', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('When the room temperature is lower than 5 °C, the valve opens; when the temperature rises to 8 °C, the valve closes.'),
            e.numeric('error', ea.STATE).withDescription('If NTC is damaged, "Er" will be on the TRV display.'),
        ],
        meta: {
            tuyaDatapoints: [
                [2, 'system_mode', tuya.valueConverterBasic.lookup({ auto: tuya.enum(0), heat: tuya.enum(1), off: tuya.enum(2) })],
                [3, 'running_state', tuya.valueConverterBasic.lookup({ heat: tuya.enum(0), idle: tuya.enum(1) })],
                [4, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
                [5, 'local_temperature', tuya.valueConverter.divideBy10],
                [7, 'child_lock', tuya.valueConverter.lockUnlock],
                [28, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
                [29, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
                [30, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
                [31, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
                [32, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
                [33, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
                [34, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
                [35, null, tuya.valueConverter.errorOrBatteryLow],
                [36, 'frost_protection', tuya.valueConverter.onOff],
                [39, 'scale_protection', tuya.valueConverter.onOff],
                [47, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration2],
            ],
        },

}

module.exports = definition;
        ```

It works for me!

cbshopper, 2024-12-18:
yes - it works, but don't ask me wy?  It is from the orignal tuya-code, a special hyperscript or javasctipt syntax

ApunktK, 2024-12-18:
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const e = exposes.presets;
> const ea = exposes.access;
> const [...]

Where should I insert this? I'm a beginner.

cbshopper, 2024-12-18:
look here: [https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_1-pairing-the-device-with-zigbee2mqtt](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_1-pairing-the-device-with-zigbee2mqtt)

Appolon20, 2024-12-19:
@cbshopper 
thanks for the hint., but it didn´t work for me.
i´ve created an external converter, pasted your code and added the converter to my configuration.yaml.
... after restart the TRV is sill in "Not Supported" state :-(

I really hope, that @Koenkk will take care of it. (if he is aware of it and have time for that integration.)

cbshopper, 2024-12-19:
have a look in your logfile and be aware about the version of your configuration.yaml. Since Version 3 you don't need to change this file, just move your new created "convertxxxx.js"  to the subfolder data/external_converters.
May be this helps! :-)
@Koenkk: I think it will help to add _TZE204_ogx8u5z6 nearby line 4725 to zigbee-herdsman-converters/devices/tuja.js

Soulfly999, 2024-12-19:
@cbshopper so it works now with your customized .js File? For me it's not working. I copied the code into a new file TZE204.js in the Z2M Folder and added in the configuration.yaml the same
`external_converters:
  - TZE204.js`

Soulfly999, 2024-12-19:
Habe es gefunden, bei musste ich es so eintrage:

`const definition = {
        zigbeeModel: ['TS0601'], // Hier die Hersteller-ID`

cbshopper, 2024-12-19:
:-)

Soulfly999, 2024-12-19:
@cbshopper is it working correct? I only can set a temperature and system mode, but it's not showing Battery state and so on

Appolon20, 2024-12-19:
> have a look in your logfile and be aware about the version of your configuration.yaml. Since Version 3 you don't need to change this file, just move your new created "convertxxxx.js" to the [...]

thanks for helping and sorry for bothering you again.
1. could you please tell me, how to figure out my version of the configuration.yaml.
2. i was not aware, that i need subfolder data/external converter, because in the guide https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition Step 2.1 it was mentioned to save the new created xxx.js next to the configuration.yaml file (for me, it means on the same level)

cbshopper, 2024-12-19:
> @cbshopper is it working correct? I only can set a temperature and system mode, but it's not showing Battery state and so on

Yes, the Battery state is the only value that not works - I've no Idear, sorry

cbshopper, 2024-12-19:
> thanks for helping and sorry for bothering you again.
>
> 1. could you please tell me, how to figure out my version of the configuration.yaml.
> 2. i was not aware, that i need subfolder data/external [...]

it depend on an entry in your configuration.yaml: "version". The documentation is not update yet, i think. I attached the 2 log-files from the last run of "update.sh", where you can see the changes
[migration-1-to-2.log](https://github.com/user-attachments/files/18199808/migration-1-to-2.log)
[migration-2-to-3.log](https://github.com/user-attachments/files/18199809/migration-2-to-3.log)
For me it was very helpful to analyze the logfile - there  you can find important hints what's going wrong.

Appolon20, 2024-12-20:
@cbshopper 
thanks very much for your feedback.
I ´ve installed version 2 of configuration.yaml but TRV still not work.

I will give up, and sell the Tuya Thermostates.

Thanks and wish all happy christmas

ApunktK, 2024-12-21:
> I had to add:
> 
> `const definition = { zigbeeModel: ['TS0601'], // Hier die Hersteller-ID`

kannst du bitte mal deine ganze js mitteilen, bei mir geht es nicht

HolgerM1, 2024-12-21:
@Koenkk Please add the support for this device. I'm having the same issue to integrate this TRV.
Thanks a lot!

janikzimmer, 2024-12-22:
Hey,

I’m really interested in seeing support for the _TZE204_ogx8u5z6 added to Zigbee2MQTT. This device would be a great addition to the project, and I believe it could benefit many users in the community, including myself.

Has anyone had the chance to test it or implement a solution? I’d be happy to help with testing or in any other way I can contribute to getting this supported. The workaround with the cbshopper code doesn’t work for me.

Thanks a lot for all your hard work on Zigbee2MQTT!

Best,
janikzimmer from Germany

ApunktK, 2024-12-22:
can you share your js file with me

evilmumi, 2024-12-22:
[ext_converter.js.txt](https://github.com/user-attachments/files/18222302/ext_converter.js.txt)
it is working - battery not shown

donotoverdriveit, 2024-12-22:
Hello, unfortunately I don't know much about it. I copied **ext_converter.js** (without .txt) into **/opt/iobroker/node_modules/zigbee-herdsman-converters/devices**. Do I need to enter anything else in **tuya.js**?

donotoverdriveit, 2024-12-22:
> [ext_converter.js.txt](https://github.com/user-attachments/files/18222302/ext_converter.js.txt) it is working - battery not shown

donotoverdriveit, 2024-12-22:
> [ext_converter.js.txt](https://github.com/user-attachments/files/18222302/ext_converter.js.txt) it is working - battery not shown

Hello, unfortunately I don't know much about it. I copied ext_converter.js (without .txt) into /opt/iobroker/node_modules/zigbee-herdsman-converters/devices. Do I need to enter anything else in tuya.js?

HolgerM1, 2024-12-22:
> > @cbshopper is it working correct? I only can set a temperature and system mode, but it's not showing Battery state and so on
> 
> Yes, the Battery state is the only value that not works - I've no Idear, sorry

In my case 3 out of four shows battery state in dashboard. Only the one I installed first doesn't. Maybe I remove and repair that one again.
[image]

yongliqi, 2024-12-23:
@HolgerM1  Normally you just need a re-interview to fix the "icon" issue. It seems that we can not get the remaining battery with a percentage. :-(

Otherwise, it is working.

Prmtlfghtr, 2024-12-23:
> [ext_converter.js.txt](https://github.com/user-attachments/files/18222302/ext_converter.js.txt) it is working - battery not shown
@evilmumi could you kindly also share your configuration.yaml? I think I've got something still wrong

Tu4r3g21, 2024-12-28:
Hi all,
Thanks to all the comments and the different options to solve this issue. I had no luck though to get these TRVs working, but I'm very sure the problem is sitting in front of the monitor. Maybe I'm doing the whole external converter topic wrong (creating the converter.js file using the HA file editor next to the zigbee2mqtt configuration.yaml + adding that .js file as an external converter via GUI), but after a Z2M restart, a new interview of these devices or even deleting and re-pairing did nothing but the same 'unsupported' notification at the device page.
So, I'll keep my fingers crossed that the solution will make it into the new version of Zigbee2Mqtt.
Cheers and a great start into 2025.

T3pco, 2024-12-31:
> > @cbshopper is it working correct? I only can set a temperature and system mode, but it's not showing Battery state and so on
> 
> Yes, the Battery state is the only value that not works - I've no Idear, sorry

Hey :)
glad to hear that it's working for you with z2m. We are testing a ZHA quirk but face some issues with adjusting the actual temp of the device. It's looking all good, but no real action (increase/decrease temp) is executed.

Can you maybe collect some debug information from
- setting a temperature value
- changing the device modes (heat, auto etc.)

It should be visible in the debug logs.

ZHA issue
https://github.com/zigpy/zha-device-handlers/issues/3628

t3pco, 2025-01-02:
I took the chance and migrated to Z2M. Except the battery it works well.
local_temperature_calibration is a bit off. The device again expects and sends multiplied values x10.

adjusted it and now it works as expected. as the device accepts at least offset of +-10 degrees, I changed that aswell.

` .withLocalTemperatureCalibration(-10, 10, 0.1, ea.STATE_SET)
[47, 'local_temperature_calibration', tuya.valueConverter.divideBy10],`

HolgerM1, 2025-01-04:
Unfortunately, after updating Core to 25.1 and Z2M to the most recent, it's not working for me anymore. I recognized the line:
 external_converters:
  - TZE204
was missing in the configuration.yaml. I added again, sadly without effect. The TRV's are not recognized anymore.

Anybody with the same issue?

yongliqi, 2025-01-04:
> Unfortunately, after updating Core to 25.1 and Z2M to the most recent, it's not working for me anymore. I recognized the line:  external_converters:   - TZE204 was missing in the configuration.yaml. [...]

Sadly, the same.

evilmumi, 2025-01-04:
anyone got schedule working?
[image]

cbshopper, 2025-01-04:
`01:00/5.0 12:00/5.0 13:00/5.0 22:00/5.0` 
worked for me -  but I don't know how to reset this field :-)
seems that you cannot leave it empty

HolgerM1, 2025-01-04:
@ErnieBall Thanks! Can confirm it's working again.

janikzimmer, 2025-01-06:
Thank you all.
I got it. But now I have another problem. One device is added without problems but my other 5 devices can not pair with my zigbee2mqtt. I enter device paring, start it on my device and zigbee2mqtt doesn't find any device.
Anyone have a solution for that problem? Is there a reset on the device possible?
Thanks

wenzelt, 2025-01-10:
I was able to copy paste into here and it worked seemlessly: 

[image]

janikzimmer, 2025-01-10:
Thanks.
Now it works but my zigbee signal is too low for every thermostat. Now I have to plan to get a bigger network :)

evilmumi, 2025-01-21:
Is it working in 2.0 without external.js?
with my .js i have problems sending from HA to it....

watn3y, 2025-01-29:
This thermostat does seem very similar to [TS0601_thermostat_3](https://github.com/Koenkk/zigbee-herdsman-converters/blob/8272fdba00459570534b6a6b262751988aded02d/src/devices/tuya.ts#L4721), however I got some more data points from the Tuya IoT cloud:

```
{
  "2": "Mode",
  "3": "Working status",
  "4": "Set temperature",
  "5": "Current temperature",
  "7": "Child lock",
  "28": "Week program",
  "29": "Week program Tuesday",
  "30": "Week program Wednesday",
  "31": "Week program Thursday",
  "32": "Week program Friday",
  "33": "Week program Saturday",
  "34": "Week program Sunday",
  "35": "Fault alarm",
  "36": "Frost protection.",
  "39": "Switch Scale",
  "47": "Temperature correction",
  "101": "Eco",
  "102": "eco temperature"
}
```
Also FYI: [ZHA seems to added native support for this device recently](https://github.com/zigpy/zha-device-handlers/pull/3682)

I'll do my best to try and open a PR adding this device but can't promise anything yet as it's my first attempt at messing with z2m.

Soulfly999, 2025-02-03:
Hey, is it now official implemented?

OpaApo, 2025-02-07:
Looking on the release note 2.1.1 https://github.com/Koenkk/zigbee2mqtt/releases it's not.

florisre, 2025-02-07:
Take a look at this implementation and feel free to comment, if I got something wrong. Without an according PR, this will never become part of the official implementation ;)

florisre, 2025-02-08:
So, support already is in the latest herdsman release, which already is in the current zigbee2mqtt dev branch. So only a matter of time now 😊

wenzelt, 2025-02-08:
Nice, awesome

ElixyrPl88, 2025-03-01:
Works with MQTT 2.1.2!
Thanks for the work.

[image]

Soulfly999, 2025-03-06:
Hey, so now it's working with the official release but it lost the battery state every time?

HolgerM1, 2025-03-06:
I do use the "workaround" from here. How can I convert to the official installation for my 6 TRV?

Soulfly999, 2025-03-08:
> I do use the "workaround" from here. How can I convert to the official installation for my 6 TRV?

As I understand it, simply rename the data/external_converters/converter.js (e.g. .bak) or delete it.

HolgerM1, 2025-03-08:
> > I do use the "workaround" from here. How can I convert to the official installation for my 6 TRV?
> 
> As I understand it, simply rename the data/external_converters/converter.js (e.g. .bak) or delete it.

Renamed the file and did a reboot. I cant not see any change, but maybe thats ok :)

evilmumi, 2025-05-08:
Now there are problems with schedules - should we open new issue?

florisre, 2025-05-09:
I assume that this is related to https://github.com/Koenkk/zigbee2mqtt/issues/25199#issuecomment-2622541543