Koenkk/zigbee2mqtt #25200: HASS Event Different to Z2M Event

D3v01dZA, 2024-12-14:
### What happened?

Using the experimental support in https://github.com/Koenkk/zigbee2mqtt/pull/24233.

I have a four button scene switch https://www.zigbee2mqtt.io/devices/TS0044.html#tuya-ts0044.

The actions for each button in Z2M are (where <button> is a button no like 1_single):
- <button>_single
- <button>_double
- <button>_hold

These all show up correctly in Z2M however in HASS the events for hold specifically do not contain the button number, only the text "hold".

So when clicking button 1 once the event in both is "1_single", when double clicking it its "1_double", but when holding it its "1_hold" in Z2M but "hold" in HASS when means I can't disambiguate which button was actually held.

### What did you expect to happen?

When holding a button I expect to know which button was held in the event.

### Zigbee2MQTT version

1.42.0 commit: 861cba6

### Adapter firmware version

7.4.2 [GA]

### Adapter

TubesZB MGM21

### Setup

x86-64 Docker

mundschenk-at, 2024-12-15:
Have a look at the other attributes (besides `event_type`.

D3v01dZA, 2024-12-15:
Ah I see the button in the attributes, not I can continue. Still it seems like this shouldn't be different based on the event type...

```
event_types:
  - 1_single
  - 1_double
  - hold
  - 2_single
  - 2_double
  - 3_single
  - 3_double
  - 4_single
  - 4_double
event_type: hold
icon: mdi:gesture-double-tap
friendly_name: Test Quadmote Action
button: "2"
```

mundschenk-at, 2024-12-15:
Well, you have just discovered why it is labelled "experimental".

D3v01dZA, 2024-12-15:
Yes - hopefully this will be seen as feedback on a new feature

mundschenk-at, 2024-12-15:
Sorry if it wasn't clear from my earlier statement: Obviously the parsing logic should (and eventually will) be expanded to cover all actions. There's just a lot of ground to cover and we are currently [discussing](https://github.com/Koenkk/zigbee2mqtt/discussions/25061) how this will be done.

D3v01dZA, 2024-12-15:
Nice! Let me close this one out then as this is obviously being looked at :)

closed 2024-12-15 by D3v01dZA