Koenkk/zigbee2mqtt #25203: [New device support]: ION Led Dimmer ID200W-ZIGB TS0601 _TZE200_ykgar0ow

wimensing, 2024-12-14:
### Link

https://www.ionindustries.com/winkel/led-zigbee-dimmer-0-3-200-watt/

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xf84477fffeea521a","nwkAddr":24334,"manufId":4098,"manufName":"_TZE200_ykgar0ow","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,4096,8,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1734192588935}

### Zigbee2MQTT version

1.42.0-2

### Comments

It appears that ION is using different hardware inside for the same ID200W-ZIGB device, because I read here (https://github.com/Koenkk/zigbee2mqtt/discussions/23955) that it can also be discovered normally. In my case it is 
My device ID200W-ZIGB with name _TZE200_ykgar0ow is different.
When I add an external converter by just adding the auto-generated definition from the Dev console, it can be added.
I can control it from Home Assistant, but the communication back from the dimmer it not received.
This is exactly the issue mentioned in https://github.com/Koenkk/zigbee2mqtt/discussions/23955

I want to use the device _TZE200_ykgar0ow  without using an external converter.
I think it should be discovered as a TS0601_dimmer_knob.

### External definition

```shell
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_ykgar0ow',
    description: 'ION Led Dimmer ID200W-ZIGB',
    extend: [light()],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I can control the dimmer from Home Assistant.
But when I use the dimmer to turn the light on or off or set the dim level, that information is not received back in Home Assistant.

wimensing, 2024-12-21:
The device _TZE200_ykgar0ow might be the same as _TZE200_4mh6tyyo.

I copied the external converter found at https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#_2-adding-your-device
Then I updated the sections "exposes" and "meta" by copying the relevant parts from model 'TS0601_dimmer_1_gang_1' found in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts

I have now this external converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_ykgar0ow',
        },
    ],
    model: 'TS0601_ion',
    vendor: 'Tuya',
    description: 'ION Led Dimmer ID200W-ZIGB',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
        exposes: [
            tuya.exposes.lightBrightnessWithMinMax(),
            tuya.exposes.countdown(),
            tuya.exposes.lightType(),
            e.power_on_behavior().withAccess(ea.STATE_SET),
            tuya.exposes.backlightModeOffNormalInverted().withAccess(ea.STATE_SET),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
                [2, 'brightness', tuya.valueConverter.scale0_254to0_1000],
                [3, 'min_brightness', tuya.valueConverter.scale0_254to0_1000],
                [4, 'light_type', tuya.valueConverter.lightType],
                [5, 'max_brightness', tuya.valueConverter.scale0_254to0_1000],
                [6, 'countdown', tuya.valueConverter.countdown],
                [14, 'power_on_behavior', tuya.valueConverter.powerOnBehavior],
                [21, 'backlight_mode', tuya.valueConverter.backlightModeOffNormalInverted],
            ],
        },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;

```

This loads in Zigbee2MTQQ and after removing and adding the dimmer I now get the data back in Home Assistant when I use the physical button of the dimmer.

NotCharmingAtAll, 2025-01-19:
Hi, I am running into the similar issues but even worse. The dimmer shows up in the device list but as "unsopported".
I want to use the ION Industries dimmer because I like the haptics of the dimmer. (indexed)
Since I am quite a noob in HA with Zigbee2MQTT, I wanted to ask you for support regarding the code above.
Where exactly did you load this code?

Thanks for your help
Matthias

wimensing, 2025-01-21:
With Zigbee2MQTT version 2 installed in Home Assistant, I placed a file called ion_led_dimmer.js in directory
/homeassistant/zigbee2mqtt/external_converters/
The file contains the external converter lines I mentioned earlier.

When Zigbee2MQTT starts it shows these lines in the logs:
...
info: 	z2m: Dimmer (0xf84477fffeea521a): Not supported (Router)
...
info: 	z2m: Loaded external converter 'ion_led_dimmer.js'.
...
info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Dimmer', payload '{"backlight_mode":null,"brightness":103,"countdown":0,"light_type":"led","linkquality":66,"max_brightness":143,"min_brightness":13,"power_on_behavior":null,"state":"OFF"}'
info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Dimmer/type', payload '{"light":"led"}'
...

So, first it says the dimmer is not supported, then it loads the external converter, and then it is loaded and published correctly to MQTT.

wimensing, 2025-01-21:
I have set the min (13) and max (143) brightness on this dimmer.
When using the physical button on the dimmer it uses these min and max settings.
But from HA you could go higher then the max.

I used Lightener (a HACS integration) to create a light button that goes from 0-100% and sets the actual dimmer 13-143.

NotCharmingAtAll, 2025-01-21:
Thanks for the explanation. This folder (/homeassistant/zigbee2mqtt/external_converters/) still works with Z2M 2.0.0?
I am asking because I found these comments in the release notes:

_- The external_converters setting is no longer used. Instead all external converters inside data/external_converters directory are now automatically loaded. Make sure to move all your external converters to this directory.
- External extensions are now loaded from data/external_extensions instead of data/extension. Make sure to rename your data/extension directory (if present) to data/external_extensions._

wimensing, 2025-01-21:
Yes, the "data/external_converters" in the release notes points to "/homeassistant/zigbee2mqtt/external_converters/" in Home Assistant.

NotCharmingAtAll, 2025-01-21:
I will give it a try ... but as far as I know, the dimmer model is already in the dev branch and should be added to the stable version with the next release. Do you know, when Z2M usually releases updates?

NotCharmingAtAll, 2025-01-21:
Thanks! It worked with your help, but only "one way".
I can control the device via HomeAssistant/Zigbee but if I move the dimmer physically or toggle the switch, this information is not fedd back into HA. Any ideas?

NotCharmingAtAll, 2025-01-21:
Strange ... disconnected the dimmer from power and reconnected ... now its working both ways.

yoricksijsling, 2025-02-16:
Looks like this device is now supported in zigbee2mqtt 2.1.0

It's added to the existing TS0601_dimmer_1_gang_1. See https://github.com/Koenkk/zigbee-herdsman-converters/pull/8564 by @redelschaap

wimensing, 2025-02-19:
Indeed, it is now supported.
I removed the device from the network, removed my external converter, and then could re-add the device again.

mozeskriebels, 2025-04-03:
I installed the ION industries LED dimmer. Zigbee2mqtt immediate detected the dimmer as a 
Zigbee model: TS0601
Zigbee Fabrikant: _TZE200_4mh6tyyo
Description: 1 gang smart dimmer

I can controle it from homeassistant. But I dont receive the right information when turning the knob. I do receive Zigbee messages but it reports the old state and not the new state of the dimmer. The light is turned off but the dimmer reports it is state is on for a full hour.
_info 2025-04-03 19:13:20z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lamp woonkamer', payload '{"backlight_mode":"off","brightness":80,"countdown":null,"last_seen":"2025-04-03T19:13:21+02:00","light_type":"led","linkquality":39,"min_brightness":75,"power_on_behavior":"previous","state":"ON"}'
info 2025-04-03 19:13:20z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lamp woonkamer/type', payload '{"light":"led"}'
info 2025-04-03 19:16:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lamp woonkamer', payload '{"backlight_mode":"off","brightness":80,"countdown":null,"last_seen":"2025-04-03T19:16:02+02:00","light_type":"led","linkquality":120,"min_brightness":75,"power_on_behavior":"previous","state":"ON"}'
info 2025-04-03 19:16:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lamp woonkamer/type', payload '{"light":"led"}'_

Do I need to install an external converter? I am running zigbee2mqtt 2.2.0-1.

redelschaap, 2025-06-03:
@mozeskriebels Still having this issue? Is the device reported the correct one, or might it be (slightly) different?

redelschaap, 2025-06-03:
Do you have a URL to the manufacturer's product page?

BorisKamp, 2025-06-23:
> Do you have a URL to the manufacturer's product page?

Hey!

I have the same problem, here is the manufacturer page for it:
https://www.ionindustries.com/winkel/led-zigbee-dimmer-0-3-200-watt/

And a screenshot of what I see in Z2MQTT when trying to connect it:

[image]
[image]

mozeskriebels, 2025-06-25:
I have this model: https://www.ionindustries.com/winkel/led-zigbee-dimmer-0-3-200-watt/ 
The product number is the same but I don't have the max/min/boost switches. 

[image]

Here is the information during and after pairing.
[image]

[image]

[image]

range-of-motion, 2026-01-16:
> I never got it to report status back.

So what did you end up doing? Accept it? Replace the dimmer?

redelschaap, 2026-01-18:
Interesting! My thought was that the configuration could be conflicting with the missing min/max/boost switches. Good to hear your version works just fine. Perhaps it's hardware related then.

Maybe @mozeskriebels and/or @BorisKamp can let us know whether their dimmer now also reports brightness correctly with the latest zigbee2mqtt version?