Koenkk/zigbee2mqtt #25207: Display stopped working and device recognized as wrong device in Z2M after upgrading firmware of TS0201

heisenberg2980, 2024-12-14:
### What happened?

I upgraded the firmware of my TS0201 in Z2M, and when the upgrade was complete the screen of the device when blank. I tried removing the batteries and putting them back after 30 minutes but the screen is still not working. I also realised the device is now recognized as LYWSD03MMC in Z2M, which is incorrect as the device is TS0201 (the picture of the device matches).

### What did you expect to happen?

Device should still work after upgrading the firmware

### How to reproduce it (minimal and precise)

Upgrade the firmware of the device. As a side note, for the upgrade to work without timing out, I had to keep pressing the button on the device again and again until the upgrade finished.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20230507

### Adapter

SMLIGHT SLZB-06

### Setup

Add-on

heisenberg2980, 2024-12-15:
According to this thread, it seems more people have had the same issue of bricking the device by applying this update: https://community.home-assistant.io/t/tuya-ts0201-temperature-sensor-ota-not-working/804442/4

Is there anything that can be done to revert the update and restore the previous firmware? And can this update be removed from Z2M to avoid other people having the same issue?

Koenkk, 2024-12-16:
Do you by any chance still have the logs of the update? Then I can check what fw was loaded and fix it.

heisenberg2980, 2024-12-16:
@Koenkk I found this on the logs before the update, is this enough?

{"battery":39,"humidity":43.2,"last_seen":"2024-12-14T09:49:12.919Z","linkquality":94,"temperature":21,"update":{"installed_version":268513281,"latest_version":285224961,"state":"available"},"update_available":null,"voltage":null}

Koenkk, 2024-12-16:
@heisenberg2980 yes, with z2m 2.0.0 (which is the latest [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) atm), you can downgrade fws. Docs: https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/docs/guide/usage/ota_updates.md#downgrade

heisenberg2980, 2024-12-16:
@Koenkk I tried, but no luck

I guest as the device "thinks" it is not a different model, it does not find the previous firmware, is it possible to force a specific fw into the device rather than asking "downgrade to the previous available fw"?

[image]

heisenberg2980, 2024-12-17:
@Koenkk I just sent you the log

Koenkk, 2024-12-20:
I removed the firmwares in https://github.com/Koenkk/zigbee-OTA/pull/629/files, unfortunately we don't have the original TS0201 firmwares so I don't think it is possible to downgrade.

closed 2024-12-21 by heisenberg2980

dragonlost, 2024-12-24:
Hello. I wanted to update my peripherals and I had the same problem as you. Their screens remained frozen (so I didn't realize they were broken). I updated 4 of them before noticing!
Did any of you manage to get back to the previous version? I don't know what to do to catch up.
I've tried changing the batteries, but I can't get a picture or a connection.

My exact model : 
temperature and humidiy sensor, Brand : NEO
model: NAS-TH02B2

Minz0r, 2025-01-24:
> I removed the firmwares in https://github.com/Koenkk/zigbee-OTA/pull/629/files, unfortunately we don't have the original TS0201 firmwares so I don't think it is possible to downgrade.

Hi, i think i found a Site with a Link to th original OTA Firmware and Custom Firmware.
[https://pvvx.github.io/TS0201_TZ3000/](url)

Celexi, 2025-01-26:
for what is worth the one I have has no screen, I got a tuya gateway to check if it really had an update and there is no update. so either that update was for the one with a display, or an entirely different one.

webgress, 2025-02-07:
> Hi, i think i found a Site with a Link to th original OTA Firmware and Custom Firmware. [https://pvvx.github.io/TS0201_TZ3000/](url)

@Minz0r any updates on this?
Did you try/succeed flashing the original FW?

I have 3 TS0201 screen devices, not round like on the site, but square. Either way, they do not report values to Z2M.
Numbers on screen seem to change but the charts have 1 straight line - no data.

Celexi, 2025-02-08:
> > > I removed the firmwares in https://github.com/Koenkk/zigbee-OTA/pull/629/files, unfortunately we don't have the original TS0201 firmwares so I don't think it is possible to downgrade.
> > 
> > 
> > Hi, i think i found a Site with a Link to th original OTA Firmware and Custom Firmware. [https://pvvx.github.io/TS0201_TZ3000/](url)
> 
> [@Minz0r](https://github.com/Minz0r) any updates on this? Did you try/succeed flashing the original FW?
> 
> I have 3 TS0201 screen devices, not round like on the site, but square. Either way, they do not report values to Z2M. Numbers on screen seem to change but the charts have 1 straight line - no data. I don't remember exactly, but I don't think I've updated them.

zigbee2mqtt doesn't automatically update anything as far I know, so if you didn't update them, this would be a different issue of it being identified wrong. did you try resetting them?

mattrj1975, 2025-02-10:
> I removed the firmwares in https://github.com/Koenkk/zigbee-OTA/pull/629/files, unfortunately we don't have the original TS0201 firmwares so I don't think it is possible to downgrade.

Then it must have accidentally crept back in to later Zigbee"MQTT updates, because I have the exact same issues after an update on a week or so old TS0201, following update 268513281 just 2 days ago.

mattrj1975, 2025-02-10:
Running Z2M 2.1.1-1