Koenkk/zigbee2mqtt #25212: Slow and not responding Network

DustinThal, 2024-12-15:
### What happened?

1. **Initial Setup**:  
   - Zigbee network with **113 devices**: 61 end devices, 52 routers.
   - Network was functioning fine before adding new devices.

2. **Problem Start**:  
   - After adding **4 new routers**, the network became slow.
   - Symptoms:  
     - Many devices do not respond or respond after a long delay (minutes or not at all).
     - Frequent **timeouts** when attempting to control devices.
     - Frequent **Failed to read state**
     - Errors: **NWK_TABLE_FULL**. <-solved

3. **Actions Taken**:  
   - Removed the newly added routers (issue persisted).  
   - Updated firmware of Zigbee coordinator stick from **20230507** to **20240710** (NWK_TABLE_FULL resolved, but issue persisted).  
   - Downgraded Zigbee2MQTT HA plugin to **1.40.2-1** in steps (no improvement).  
   - Installed the latest Zigbee2MQTT Edge version (no improvement).

4. **Current State**:  
   - Network remains slow.
   - Many devices fail to respond to commands in time, with **timeouts** occurring frequently.

### What did you expect to happen?

normal behavior of the system

### How to reproduce it (minimal and precise)

Hardware:
DELL Optiplex 3050 connected(wired) to FritzBox7590 (OS 8.00)
UZG-01 FW-Rev 20240707 HW-rev ESP32-D0WD-V3 with CC2652P7 FW-rev 20240710 connected(wired) to FritzBox
Home Assistant Core 2024.12.3 Super 2024.11.4 OS 14.0 Front 20241127.8

### Zigbee2MQTT version

1.42.0-dev commit: e234f42

### Adapter firmware version

20240710

### Adapter

UZG-01

### Setup

Add-on on Home Assistant OS on Dell Optiplex 3050

### Debug log

[log.log](https://github.com/user-attachments/files/18139425/log.log)

IceColdCrash, 2024-12-17:
Same here, never had this lag before the update, some work as expected but many take sometimes 15 or 20 seconds to respond

Rozenlaan, 2024-12-18:
Same here Frequent timeouts when attempting to control devices.

FuturamaPhill, 2024-12-23:
Same issue here, never had these sort of issues before but now devices are very slow to respond and getting a lot of timeouts

IceColdCrash, 2024-12-23:
Got mine fixed, after changing so many things,
I've got 2 Z2M networks, one on the garage and another at home, I had slzb-06p7 at home and slzb-06 in the garage,
Managed to swap them without re-pairing everything and issue was still going not as bad but still not great,
I was getting a lot of "ZCL" errors and as my Zigbee channel is on 25 and wifi on channel 1 never thought about neighbours interference. 
After so many things changed I decided to investigate neighbours wifi (I live in a block of apartments), found an hp printer wifi, now this will be funny, I had somehow to see how I would discard this but I don't know who it belongs to, searched Google and saw that particular printer has a default wifi password, managed to connect myself to it but no option to turn it's wifi off so I tried to print something on it (the instruction manual) so the owner would think it was going crazy and disconnect it, wouldn't let me print so I tried the scan option, that worked, 2 minutes after it started scanning owner must have heard and it's been off since then, don't know if this was the issue, I know that everything is now running fast and perfect as usual but changed so many things  that I just don't know
All this to say, you guys better check for interference with Wi-Fi
Merry Christmas 👍

Teeseeone, 2025-01-09:
Suddenly my zigbee network is acting as others above, up to 20 sec delay. Not too many days ago everything worked in an instant. Was really happy with the speed and communication. Not really sure what has changed in my setup. Still running HA 2025.1.1 and z2m 1.42.0-2.

IceColdCrash, 2025-01-09:
> Suddenly my zigbee network is acting as others above, up to 20 sec delay. Not too many days ago everything worked in an instant. Was really happy with the speed and communication. Not really sure what has changed in my setup. Still running HA 2025.1.1 and z2m 1.42.0-2.
> 
> edit: Running SLZB-06P7 on current core firmware version: v2.5.8

Make sure it's not a neighbour's wi-fi, in my case it always a damn HP deskjet printer from one of the neighbour's
I've gut everything updated just like you, same versions and all is perfect

DustinThal, 2025-01-26:
again, one instance crashed completely, weird is, that configuration from z2m and backup of coordinator has my devices, but z2m ui and HA does not, unfortunately I lost my logs from time where it happens ....but this time I only have to rejoin about 30 devices, and not around 100.
next step is to set up a syslog -.-

closed 2025-04-04 by github-actions[bot]