Koenkk/zigbee2mqtt #25216: [New device support]:

simone-78, 2024-12-15:
### Link

https://it.aliexpress.com/item/1005005276875340.html?spm=a2g0o.detail.similar_items.5.3b6dxd0Txd0TH3&utparam-url=scene%3Aimage_search%7Cquery_from%3Aapp_pdp_sold_out&pdp_npi=4%40dis%21EUR%2179.32%2124.59%21%21%21590.18%21182.96%21%40211b653717342613440625799e5b86%2112000032470267523%21sea%21IT%21761135482%21X#nav-review

### Database entry

{"id":18,"type":"Router","ieeeAddr":"0xa4c138fcf8f91afa","nwkAddr":49060,"manufId":4417,"manufName":"_TZE204_navtwmd0","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1734258868981}

### Zigbee2MQTT version

1.42.0

### Comments

Goodmorning, I'm trying to get a working converter for this new thermostat. I can't find anything similar to clone to get basic functionality. I apologize, but unfortunately I don't have the skills to make this work. Could you help me?

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_navtwmd0',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

the external definition only provides link quality, No further functionality

jemontejo, 2025-01-20:
I only have the thermometer version...

Save code below as '_TZE204_navtwmd0.js' inside '/homeassistant/zigbee2mqtt/external_converters/' folder

-------------

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');

const tz = require('zigbee-herdsman-converters/converters/toZigbee');

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_navtwmd0',
        },
    ],
    model: 'TS0601',
    vendor: 'Tuya',
    description: 'Tuya Smart Zigbee Thermometer',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.temperature() // Here you should put all functionality that your device exposes
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'temperature', tuya.valueConverter.divideBy10]
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;

TarasRybak, 2025-01-22:
works. thank you

[image]

TarasRybak, 2025-01-22:
but 'z2m state' has additional parameters, can they be added to the converter?

       "0xa4c138f7121f75eb": {
            "date_code": "",
            "definition": {
                "description": "Tuya Smart Zigbee Thermometer",
                "exposes": [
                    {
                        "access": 1,
                        "description": "Measured temperature value",
                        "label": "Temperature",
                        "name": "temperature",
                        "property": "temperature",
                        "type": "numeric",
                        "unit": "°C"
                    },
                    {
                        "access": 1,
                        "category": "diagnostic",
                        "description": "Link quality (signal strength)",
                        "label": "Linkquality",
                        "name": "linkquality",
                        "property": "linkquality",
                        "type": "numeric",
                        "unit": "lqi",
                        "value_max": 255,
                        "value_min": 0
                    }
                ],
                "model": "TS0601",
                "options": [
                    {
                        "access": 2,
                        "description": "Calibrates the temperature value (absolute offset), takes into effect on next report of device.",
                        "label": "Temperature calibration",
                        "name": "temperature_calibration",
                        "property": "temperature_calibration",
                        "type": "numeric"
                    },
                    {
                        "access": 2,
                        "description": "Number of digits after decimal point for temperature, takes into effect on next report of device. This option can only decrease the precision, not increase it.",
                        "label": "Temperature precision",
                        "name": "temperature_precision",
                        "property": "temperature_precision",
                        "type": "numeric",
                        "value_max": 3,
                        "value_min": 0
                    }
                ],
                "supports_ota": false,
                "vendor": "Tuya"
            },
            "disabled": false,
            "endpoints": {
                "1": {
                    "bindings": [],
                    "clusters": {
                        "input": [
                            "genGroups",
                            "genScenes",
                            "manuSpecificTuya",
                            "genBasic"
                        ],
                        "output": [
                            "genOta",
                            "genTime"
                        ]
                    },
                    "configured_reportings": [],
                    "scenes": []
                },
                "242": {
                    "bindings": [],
                    "clusters": {
                        "input": [],
                        "output": [
                            "greenPower"
                        ]
                    },
                    "configured_reportings": [],
                    "scenes": []
                }
            },
            "friendly_name": "11111111",
            "ieee_address": "0xa4c138f7121f75eb",
            "interview_completed": true,
            "interviewing": false,
            "manufacturer": "_TZE204_navtwmd0",
            "model_id": "TS0601",
            "network_address": 63626,
            "power_source": "Mains (single phase)",
            "supported": true,
            "type": "Router"
        }
    },

jemontejo, 2025-02-07:
if you happen to know the Data point ID for temperature_calibration and temperature_precision then we can add them...unfortunately, i do not have tuya gateway to find this out...

https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

AxelAngelo, 2025-04-12:
Thank you, thank you, thank you !

closed 2025-06-20 by github-actions[bot]