Koenkk/zigbee2mqtt #25220: Doesn't start after generating new network key

bartkummel, 2024-12-15:
### What happened?

The [documentation](https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html) states:

> Set `network_key: GENERATE` to let Zigbee2MQTT generate a new random key on the first start. The `configuration.yml` gets updated with the new key. Changing the network_key requires repairing of all devices.

I had no devices paired yet, so it seemed a good moment to do this. So I changed the values of `network_key` and `ext_pan_id` to `GENERATE` and restarted. Then it didn't want to restart anymore, due to these errors:

```
[2024-12-14 20:25:39] error: 	zh:adapter:zstack:manager: Configuration is not consistent with adapter state/backup!
[2024-12-14 20:25:39] error: 	zh:adapter:zstack:manager: - PAN ID: configured=****, adapter=****
[2024-12-14 20:25:39] error: 	zh:adapter:zstack:manager: - Extended PAN ID: configured=****, adapter=****
[2024-12-14 20:25:39] error: 	zh:adapter:zstack:manager: - Network Key: configured=****, adapter=****
[2024-12-14 20:25:39] error: 	zh:adapter:zstack:manager: - Channel List: configured=11, adapter=11
[2024-12-14 20:25:39] error: 	zh:adapter:zstack:manager: Please update configuration to prevent further issues.
[2024-12-14 20:25:39] error: 	zh:adapter:zstack:manager: If you wish to re-commission your network, please remove coordinator backup at /config/zigbee2mqtt/coordinator_backup.json.
[2024-12-14 20:25:39] error: 	zh:adapter:zstack:manager: Re-commissioning your network will require re-pairing of all devices!
[2024-12-14 20:25:39] error: 	z2m: Error while starting zigbee-herdsman
[2024-12-14 20:25:39] error: 	z2m: Failed to start zigbee
[2024-12-14 20:25:39] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-12-14 20:25:39] error: 	z2m: Exiting...
[2024-12-14 20:25:39] error: 	z2m: Error: startup failed - configuration-adapter mismatch - see logs above for more information
    at ZnpAdapterManager.determineStrategy (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:251:35)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:62:26)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)
```
(Note that I redacted the actual keys.)

This was not a very pleasant way to start with Zigbee2MQTT. Correcting the problem was also not very easy. Of course it is very handy that the logs show the values of the adapter, so that I could edit the config again to match the adapter. However, in the config file, I need to enter it as an array, while the logs show the keys as string. That's inconvenient and it took me a few attempts to figure out the correct format.

### What did you expect to happen?

The documentation should warn for this. Also, recovering should be easier by making sure that the keys from the logs can be copy/pasted right into the configuration.

### How to reproduce it (minimal and precise)

Already described under "What happened"

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

Add-on on HomeAssistant OS.

Koenkk, 2024-12-16:
Removing the `data/coordinator-backup.json` should fix this issue (it's a safety mechanism to prevent people from losing their networks)

bartkummel, 2024-12-16:
I'd suggest to add that info to the docs.

Koenkk, 2024-12-17:
Makes sense! Would you mind making a PR by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/edit/master/docs/guide/installation/20_zigbee2mqtt-fails-to-start.md)?

bartkummel, 2024-12-18:
https://github.com/Koenkk/zigbee2mqtt.io/pull/3339

andrazek, 2024-12-30:
Today, I finally found time to secure my zigbee netowrk by setting GENERATE for pan_id, ext_pan_id, and network_key and restarting. However, I have a question: why is the network_key not visible in the UI (Settings > Advanced), while the other two keys are displayed? All three keys are present in configuration.yaml. I only noticed this after re-pairing half of my network. Now I’m unsure whether the generated network_key is actually being used or if it’s still the default one (I’ve been a Z2M user for a long time). I'm on the last version of Zigbee2MQTT, and my Dongle Plus P has the latest FW.

closed 2025-03-17 by github-actions[bot]