Koenkk/zigbee2mqtt #25229: Ikea E2013 Parasoll not recognized

pumuqui, 2024-12-16:
### What happened?

Tried to pair Ikea E2013 Parasoll device resulted in Zigbee2MQTT stop. Seems to be a problem in interviewing the device.
Ikea E2134 Vallhorn pairs without problems.
These are my first attempts with HA and z2m and I wondered that E2013 is not working since it is on the supported device list.

### What did you expect to happen?

The device should be paired.

### How to reproduce it (minimal and precise)

- Allow pairing in Zigbee2MQTT.
- Press pairing button 4 times on device.

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

8.0.2 [GA]

### Adapter

ZBDongle-E

### Setup

Docker image https://github.com/Koenkk/zigbee2mqtt

### Debug log

[log.log](https://github.com/user-attachments/files/18154717/log.log)

best-74, 2024-12-19:
Same issue here. Running latest z2m version.

best-74, 2024-12-21:
Hmm, i just tried another device, for me, both test devices shows up as unsupported...

closed 2025-04-14 by github-actions[bot]