Koenkk/zigbee2mqtt #25239: MOES Carbon Monoxide Alarm Model: HS-720ES

EmanueleGuidetti, 2024-12-17:
### Link

https://it.aliexpress.com/item/1005007205819075.html?spm=a2g0o.order_list.order_list_main.9.6d6d3696wMCRvZ&gatewayAdapt=glo2ita

### Database entry

{"id":140,"type":"EndDevice","ieeeAddr":"0xa4c138ba38a3ef5b","nwkAddr":19990,"manufId":4417,"manufName":"_TZE284_rjxqso4a","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"+\u0001\u0000\u0000e,\u0001\u0000\u0000\u0012","65506":56,"65508":1,"appVersion":77,"stackVersion":0,"dateCode":"","modelId":"TS0601","manufacturerName":"_TZE284_rjxqso4a","powerSource":3,"zclVersion":3,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1734421755957}

### Zigbee2MQTT version

1.42.0

### Comments

Hi,
I've CO sensor MOES HS-730ES.
It's seems the same that MOES ZC-HM (https://www.zigbee2mqtt.io/devices/ZC-HM.html) and HEIMAN HS-730ES (https://www.zigbee2mqtt.io/devices/HS-720ES.html) 

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_rjxqso4a',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I'm not able to define extended definition and support device.
Can you help me?
Thanks!

luckbert, 2024-12-17:
Hello, 
Unfortunately I have the same problem and cannot find a solution. 
Greetings Luxi

EmanueleGuidetti, 2024-12-18:
I tried duplicating the configuration of MOES ZC-HM (from here:  and only changed the fingerprint. Then, I added the configuration to "external_converters," but the device still isn't recognized.

This is the external definition I've try:

```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE284_rjxqso4a']),
    model: 'HS-720ES',
    vendor: 'Moes',
    description: 'Carbon Monoxide Alarm',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        /* DP  1 */ e.carbon_monoxide(),
        /* DP  2 */ e.co(),
        /* DP  9 */ tuya.exposes.selfTestResult(),
        /* DP 15 */ e.battery(),
        /* DP 16 */ tuya.exposes.silence()
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'carbon_monoxide', tuya.valueConverter.trueFalse0],
            [2, 'co', tuya.valueConverter.raw],
            [9, 'self_test_result', tuya.valueConverter.selfTestResult],
            [15, 'battery', tuya.valueConverter.raw],
            [16, 'silence', tuya.valueConverter.raw]
        ]
    }
};

module.exports = definition;
```

EmanueleGuidetti, 2024-12-18:
I have remove external_converter and add manufacturer name in original definition:

`/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/moes.js`

In the file i have found the definition of MOES ZC-HM and add  '_TZE284_rjxqso4a' in the fingerprint:

`fingerprint: tuya.fingerprint('TS0601', ['_TZE200_hr0tdd47', '_TZE200_rjxqso4a', '_TZE284_rjxqso4a']),`

Restart Ziggbee2mqtt and now device is fully supported!

Tested:
  'carbon_monoxide' OK!
  'co', OK!
  'self_test_result', OK!
  'battery', I'm to wait some hours...
  'silence', I've to understood how it's work...

luckbert, 2024-12-18:
Hello EmanueleGuidetti,
Unfortunately, I didn't quite understand what exactly you did. 
Can you explain it to me in more detail?
Regards

EmanueleGuidetti, 2024-12-18:
You have to find your /zigbee2mqtt folder installation and Inside, you will find the path to the specified file.
My folder may be different because I use LCX container for Zigbee2mqtt installation.
I hope this can help you

luckbert, 2024-12-18:
Unfortunately it doesn't work for me because I run z2M in an addon and I can't access the files. I apparently have to wait until it is adopted in an update. 
Regards

aubierr, 2024-12-18:
Hello EmanueleGuidetti,

I just got 2 HS-720ES detectors, I added the file below.
I had to remove constants from the file above for it to be recognized but it doesn't work. 
No information comes back on Z2M

Regards

```
// const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
// const reporting = require('zigbee-herdsman-converters/lib/reporting');
// const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_hr0tdd47', '_TZE200_rjxqso4a', '_TZE284_rjxqso4a']),
    model: 'HS-720ES',
    vendor: 'Moes',
    description: 'Carbon Monoxide Alarm',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        /* DP  1 */ e.carbon_monoxide(),
        /* DP  2 */ e.co(),
        /* DP  9 */ tuya.exposes.selfTestResult(),
        /* DP 15 */ e.battery(),
        /* DP 16 */ tuya.exposes.silence()
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'carbon_monoxide', tuya.valueConverter.trueFalse0],
            [2, 'co', tuya.valueConverter.raw],
            [9, 'self_test_result', tuya.valueConverter.selfTestResult],
            [15, 'battery', tuya.valueConverter.raw],
            [16, 'silence', tuya.valueConverter.raw]
        ]
    }
};

module.exports = definition;
```

aubierr, 2024-12-20:
I found the solution. I run Zigbee2MQTT as an addon also on Home Assistant. To make a manual file you have to add the name of the file that you add to the configuration.yaml file of the Zigbee2MQTT folder

Example :
```yaml
external_converters: 
  - moes_perso.js
```
Then you have to create the file of the same name, also in the Zigbee2MQTT directory

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE284_rjxqso4a',
        },
    ],
    model: 'HS-720ES',
    vendor: 'Moes',
    description: 'Carbon monoxide alarm',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        /* DP  1 */ e.carbon_monoxide(),
        /* DP  2 */ e.co(),
        /* DP  9 */ tuya.exposes.selfTestResult(),
        /* DP 15 */ e.battery(),
        /* DP 16 */ tuya.exposes.silence()
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'carbon_monoxide', tuya.valueConverter.trueFalse0],
            [2, 'co', tuya.valueConverter.raw],
            [9, 'self_test_result', tuya.valueConverter.selfTestResult],
            [15, 'battery', tuya.valueConverter.raw],
            [16, 'silence', tuya.valueConverter.raw]
        ]
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
```
Finally, you have to associate the devices again

luckbert, 2024-12-20:
Thank you very much!
Great job!!!
It worked!!!!

Justblair, 2024-12-21:
I have added a pull request to the Herdsman repository to modify the signature as per https://github.com/Koenkk/zigbee2mqtt/issues/25239#issuecomment-2551007332 Hopefully the new model will be supported soon.

yboujraf, 2024-12-21:
Dear @aubierr 

I tried on Z2M addons

Devices :

[image]

About :

[image]

Terminal : Z2M ADDONS

[image]

I created the : moes.js file with your code on same directory of configuration.yaml

Question : 

Do I need to add : external_converters: 

[image]

or through TERMINAL : 

in configuration.yaml ?

Even tried on both cases :

I got this

`warning 2024-12-21 23:25:49z2m: Device '0xa4c1384ad4d3391c' with Zigbee model 'TS0601' and manufacturer name '_TZE284_rjxqso4a' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html`

----

FYI, on "Smart Life" apps, it is detected and get notifications, Level, ..;

[image]

Best Regards

yboujraf, 2024-12-21:
Dear @aubierr 

I reply to myself : 

1. ADDONS Terminal & SSH : 

[image]

2. OPEN WEB UI

[image]

3. Create the External definition: 

[image]

4. Zigbee2MQTT - "OPEN WEB UI"
- Select "wheele" > Settings > External Converters > + > add the filename "moes.js" created in Zigbee2mqtt directory > Submit

[image]

Then when I try to PAIR

5. About : 

[image]

6. Expose : 

[image]

7. Dashboard : 

[image]

But missing "Power" : NA  but the value is available 

[image]

Best Regards,

scorfman, 2024-12-24:
I added external support for this device I just set up and report it works great. All values reported.

Gorgue, 2025-01-04:
I have this same problem, I created the file with the code and added it as an external_converter, after restarting I was able to connect the device, my problem now is that somes values ​​come to null. 
[image]

mckar, 2025-01-15:
@EmanueleGuidetti I think you can close this issue as it was added to Zigbee Herdsman Converter Version
21.7.0 - see [#8496](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8496) and Z2M 2.0.0 includes 21.12.0. I can confirm I was able to add it without modifications.

closed 2025-01-19 by EmanueleGuidetti

danielrosehill, 2025-02-06:
Facing the same challenge ... refuses to integrate with Zigbee 2 MQTT.

Anybody know definitively what the pairing mode operation is? 

[image]

nimdaxxx, 2025-04-02:
> Facing the same challenge ... refuses to integrate with Zigbee 2 MQTT.
> 
> Anybody know definitively what the pairing mode operation is?
> 
> [image]

I was looking for the same answer :) I just received the detector, but as soon as I inserted the batteries, it didn’t automatically enter pairing mode. So I tried searching online without success—I even asked ChatGPT :) In the end, I solved it the old-fashioned way: trial and error :) You need to quickly press the test button—I didn’t count how many times, but probably more than 3. Live Long and Prosper

charlesrg, 2025-10-16:
Have you guys tested this device ? I just tried mine with a candle under it enclosed on cardboard, and the devices kept saying 0 CO.

aubierr, 2025-10-16:
Same problem for me