Koenkk/zigbee2mqtt #25241: IKEA LED1923R5 (GU10 rgbw)and LED2111G6 (E14 rgbw) freeze controlled by z2m and RDM004 (Hue Wall Switch Module)

SkyhawkXava, 2024-12-17:
### What happened?

Hello,

I am using the
* Signify Hue Wall Switch Module (RDM004)

to directly bind with the following:
' IKEA LED1923R5 (TRADFRI bulb GU10, color/white spectrum, 345/380 lm), firmware version 1.0.38
* IKEA LED2111G6 (TRADFRI bulb E14, color/white spectrum, globe, opal, 806 lm), firmware version 1.0.38

I have the following problem: When I turn off the above-mentioned lamps via Z2M or HA and then try to turn the lamps back on using the Hue Wall Switch, the lamps remain off. Afterward, the lamps cannot be turned on by any means. (Although the on/off status changes according to Z2M, the lamps remain off. Only after disconnecting the lamps from the power supply do they start working again.)

Apparently, this behavior can also be reproduced with ZHA as the Zigbee coordinator.
(See: https://community.home-assistant.io/t/philips-hue-wall-module-issue-with-ikea-gu10-bulbs/802481)

With all other Zigbee light bulbs, this issue does not occur.

### What did you expect to happen?

Turn on the light with Hue Wall Switch Module after turning off with using z2m.

### How to reproduce it (minimal and precise)

1.) Turn off IKEA LED1923R5 (GU10 rgbw) or LED2111G6 (E14 rgbw) with z2m.
2.) Turn on the light again with hue wall switch module.
3.) The light is freezed. (Until it was disconnected from the power again.)

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.2.1.0 build 144

### Adapter

EZSP v10

### Setup

Add-on on Home Assistant OS

SkyhawkXava, 2025-02-23:
No Solutoon setzt. Is there a possible way or workaround in z2m?

SkyhawkXava, 2025-05-04:
Unfortunately, the problem still persists. Is there a way to work around it? Has anyone listened in on the commands (via sniffer) of the Hue Wall Switch Module?

christophhaase, 2025-05-24:
I have exactly the same problem with Ikea LED2005R5 (TRADFRI bulb GU10, white spectrum, 345/380 lm) with firmware 3.0.21.

ChristianPorsche, 2025-07-18:
I have exactly the same problem as well

closed 2025-11-29 by github-actions[bot]