Koenkk/zigbee2mqtt #25242: Philips/Signify Hue Wall Switch Module (model 929003017102, RDM004) with direct binding => no dimming

SkyhawkXava, 2024-12-17:
### What happened?

Zigbee2MQTT version: 1.42.0-2 (via Home Assistant with SkyConnect)
Hello,

I have several Signify Hue wall switch modules (model 929003017102, RDM004) directly bound to various light bulbs. Turning the lights on and off works so far, but the light bulbs (IKEA/Tradfri and Müller Licht/Tint) cannot be dimmed. Of course, the switch's "device mode" option has been set to "single_push_button".

The Hue Wall Switch Module is directly bound to the “LevelCtrl” and “OnOff” clusters for the respective Zigbee groups.

Under “Directly control Zigbee groups” for Philips 929003017102 in the Z2M device database, it states that dimming should be possible with direct binding if the “LevelCtrl” cluster is bound to the respective group.

### What did you expect to happen?

Dimming Lights via ZigBee-Group by using Hue Wall Switch Module (long press/hold the push button).

### How to reproduce it (minimal and precise)

1.) Create a group of lights/bulbs.
2.) Pair a "Hue Wall Switch Module" to z2m.
3.) Set option "device mode" of the "Hue Wall Switch Module" to "single_push_button".
4.) Bind Cluster "groups" to Coordinator.
5.) Add the "Hue Wall Switch Module" to the above group.
6.) Unbind Clusters "LevelCtrl", "OnOff" and "manuSpecificPhilips" to Coordinator.
7.) Bind Clusters "LevelCtrl" and "OnOff" to above group.
8.) Push and hold the push button, which ist connected to the "Hue Wall Switch Module".

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.2.1.0 build 144

### Adapter

EZSP v10

### Setup

Add-on on Home Assistant OS

S-blok, 2024-12-26:
Same problem over here. 
Using the Hue wall switch module via SLZB-06M and z2m. Followed the steps on https://www.zigbee2mqtt.io/devices/929003017102.html#philips-929003017102. The on/off works, the dimming won't work.

When I hold down the push button, the lamp turns on or off. It doesn't dim

tsloms, 2025-02-07:
> Same problem over here. Using the Hue wall switch module via SLZB-06M and z2m. Followed the steps on https://www.zigbee2mqtt.io/devices/929003017102.html#philips-929003017102. The on/off works, the dimming won't work.
> 
> When I hold down the push button, the lamp turns on or off. It doesn't dim

Same here. Exactly same layout. 
Seems that the switch activates as soon as the button is closed (rather than released). Is there a way to fix that?

philipgo, 2025-02-25:
According to the Zigbee2MQTT docs this should work. Can anyome confirm it worked in the past?

tsloms, 2025-02-25:
> According to the Zigbee2MQTT docs this should work. Can anyome confirm it worked in the past?
No, I can't confirm that, as I only recently tried it out.

SkyhawkXava, 2025-02-25:
On the one hand, the hardware revisions RDM001 and RDM004 behave differently. However, this also seems to depend heavily on the installed firmware version. Even with RDM001 running the latest firmware version, there still appear to be issues.

See also issue #19941 (Can´t add Hue wall switch to Zigbee group).

philipgo, 2025-02-25:
Mine is RDM004 and dimming does not work. With RDM001 it appears to be working only with older firmware. 

Can the hardware version be found on the device without pairing it?

tsloms, 2025-02-26:
I've got the RDM004 version. Firmware version 2.60.16. Model says 929003017102.

philipgo, 2025-02-27:
I bought a used RDM001 (model 9290030171). Unfortunately it came with firmware 2.77.35 and shows the same problems.

janhenk102, 2025-03-30:
Issue apparently not resolved yet. Today I migrated my Hue Wall Switch to Z2M and I wanted it to bind it to a group. Switching on/off works fine, but dimming (long press) doesn’t. I use RDM001 and a Smlight SLZB-06M.

On/off toggle is on press, it doesn’t “wait” for de long press.

tsloms, 2025-05-25:
Is there any update on this, whether and how this can possibly be fixed?

niklasschloegel, 2025-07-25:
I have the same issue, but adding the wall switch module to the group does not work for me.

trollix, 2025-08-17:
I have the same issue, but adding the wall switch module to the group does not work for me.

Bastian007, 2026-10-13:
problem persists