Koenkk/zigbee2mqtt #25243: [New device support]: Müller Licht tint Smart light Cano ceiling light silver white+color

SkyhawkXava, 2024-12-17:
### Link

https://www.mueller-licht.de/produktinformationen/artikel/?q=404123

### Database entry

{"id":34,"type":"Router","ieeeAddr":"0xa4c13800aa46357a","nwkAddr":61360,"manufId":4635,"manufName":"MLI","powerSource":"Mains (single phase)","modelId":"Ceiling light","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,4111,65376],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"appVersion":0,"hwVersion":0}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":555}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x000d6f000dc5596f","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x000d6f000dc5596f","endpointID":1}],"configuredReportings":[{"cluster":8,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":1},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"MDPANL501","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1734434488005}

### Zigbee2MQTT version

1.42.0

### Comments

There is also an identical model in black. (https://hey-tint.de/products/tint-smarte-deckenleuchte-cano-black-white-color?srsltid=AfmBOor4TTbSZORUsKGYo960LnFC1pse_4WCli-tgPPmXg9s5ItTBLBS)

### External definition

```shell
const definition = {
    zigbeeModel: ['Ceiling light'],
    model: 'Ceiling light',
    vendor: 'MLI',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[153,555]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};
```

### What does/doesn't work with the external definition?

Seems to work so far out of the box in z2m so far.
It just doesn't appear in the OTA area.

closed 2025-02-23 by github-actions[bot]

SkyhawkXava, 2025-02-23:
The device has unfortunately not been added yet.

SkyhawkXava, 2025-03-07:
@Koenkk: Is anything still missing for the ceiling lamp to be added to z2m?

Koenkk, 2025-03-08:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/muller_licht.ts)?

SkyhawkXava, 2025-04-13:
Yeah, sure, I just did that.

[pull request #9172: Add new device in muller_licht.ts: Tint Cano ceiling light 404122/404123](https://github.com/Koenkk/zigbee-herdsman-converters/pull/9172)