Koenkk/zigbee2mqtt #25251: Rovito-Z’s backlight can’t be turned off

GilDev, 2024-12-17:
### What happened?

When trying to turn off this light, the main light turns off but not the color backlight. I found no way to turn it off.

Interestingly, it works using a binded switch on the On/Off cluster (using a [Polynhome command module](https://www.polynhome.io/produit/nano-commande-sans-fil-2-voies/), not yet supported by Z2M).

At one point, playing with the commands, I even made the light crash somehow, had to fully power cycle the device for it to work again, otherwise all commands failed to send from Z2M.

### How to reproduce it (minimal and precise)

Turn on the light any way you want (the backlight is also automatically enabled apparently), then try to turn it off. Only the main light turns off, the backlight stays on.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.4.4.0 build 0

### Adapter

Sonoff ZBDongle-E (EZSP v13)

### Setup

Add-on on HAOS on x86 Proxmox server

### Debug log

[log.log](https://github.com/user-attachments/files/18170420/log.log)

GilDev, 2024-12-27:
Interesting that no one is having this issue. It’s the light I have the most problems with. Still can’t turn it off via Z2M, I may have to return it sadly…

GilDev, 2025-01-15:
Sending raw commands to cluster On/Off of endpoint 1 works as expected, weird but reassuring. Now I just got to find a way to fix the converter, which does not seem very obvious, wish me luck!

```
TOGGLE RAW
info 2025-01-15 21:09:32zhc:tz: Invoked '6.2' with payload '{}'
info 2025-01-15 21:09:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Plafonnier salle d'eau', payload '{"brightness":239,"color":{"h":191,"hue":191,"s":10,"saturation":10,"x":0.3119,"y":0.3277},"color_mode":"xy","color_temp":153,"color_temp_startup":65535,"effect":null,"linkquality":164,"power_on_behavior":"previous","state":"OFF"}'

TOGGLE RAW
info 2025-01-15 21:09:35zhc:tz: Invoked '6.2' with payload '{}'
info 2025-01-15 21:09:35z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Plafonnier salle d'eau', payload '{"brightness":239,"color":{"h":191,"hue":191,"s":10,"saturation":10,"x":0.3119,"y":0.3277},"color_mode":"xy","color_temp":153,"color_temp_startup":65535,"effect":null,"linkquality":160,"power_on_behavior":"previous","state":"ON"}'

TOGGLE Z2M
info 2025-01-15 21:09:38z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Plafonnier salle d'eau', payload '{"brightness":239,"color":{"h":191,"hue":191,"s":10,"saturation":10,"x":0.3119,"y":0.3277},"color_mode":"xy","color_temp":153,"color_temp_startup":65535,"effect":null,"linkquality":164,"power_on_behavior":"previous","state":"OFF"}'

TOGGLE Z2M
info 2025-01-15 21:09:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Plafonnier salle d'eau', payload '{"brightness":239,"color":{"h":191,"hue":191,"s":10,"saturation":10,"x":0.3119,"y":0.3277},"color_mode":"xy","color_temp":153,"color_temp_startup":65535,"effect":null,"linkquality":164,"power_on_behavior":"previous","state":"ON"}'

kianusch, 2025-01-28:
With Z2M version 2.0.0, it seems to work as expected:

Sending {"state": "OFF"} turns off both the main light and the "ambiance" light.
Similarly, sending {"state": "ON"} turns on both the main light and the "ambiance" light.

Sending {"brightness": 0} turns off the main light, but the "ambiance" light stays on. Since the brightness of the ambiance light cannot be controlled.

Also the ambiance light can not be turned off—as long as the main light remains on.

GilDev, 2025-02-08:
Damn it, I found the bug!

After debugging things with @kianusch in #20454, I found what was happening. If you set the “Transition” parameter to anything (even 0) in the “Settings (specific)” of the device, then the color backlight never turns off!

As soon as you empty the field, it starts working again. Maybe the converter could be adapted: it could send a new on/off command without the transition parameter after the transition time has elapsed in order to turn off the backlight while still keeping the transition for the main light.

kianusch, 2025-02-09:
Is the transition parameter handled by the device itself? If this is the case, I think it would not be that easy to fix it the way you propose.

It would probably be easier to adapt your home automation workflow.

Check if the transition time is requested and greater than 0, and if the final brightness is set to 0.
In this case:

Send the command (including the desired transition time).

Wait for the transition time period.

Send the command again without the transition time.

(This is pretty much what you described, but letting your home automation handle it.)

GilDev, 2025-02-09:
Yeah indeed that’s what I tried to describe but I would have thought that it should have been made in Z2M, so that the end user just has a basic control that has the attended behavior. Granted it’s probably more complicated to implement this way, but external converters can be scripted so it might be possible, I’ll see if I can try doing it some point. 🙂

GilDev, 2025-04-11:
Haven’t got the chance to look at a fix of the converter yet.

closed 2025-06-19 by github-actions[bot]