Koenkk/zigbee2mqtt #25255: ED-10014 correct pairing but Failed to register group

Anpro88, 2024-12-18:
### What happened?

I got 67 Zigbee devices online including 10 wall buttons, 9 EcoDim Zigbee 8 button wall switch - white ED-10014 and a Sunricher Zigbee wall controller - 8 button VES-ZB-WAL-012.
All come online and report an LQI and a battery state - about 6/10 work normal but the rest produces the following logs - sometimes it even happens, that three of four buttons work:

Error 2024-12-18 09:25:14zh:ember: Failed to register group '48689' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:25:20zh:ember: Failed to register group '2305' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:25:23zh:ember: Failed to register group '2321' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:25:28zh:ember: Failed to register group '2337' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:25:34zh:ember: Failed to register group '2353' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:25:43zh:ember: Failed to register group '24577' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:25:47zh:ember: Failed to register group '24593' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:25:53zh:ember: Failed to register group '24609' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:25:59zh:ember: Failed to register group '24625' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:26:13zh:ember: Failed to register group '27905' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:26:18zh:ember: Failed to register group '27921' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:26:24zh:ember: Failed to register group '27937' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:26:28zh:ember: Failed to register group '27953' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:29:01zh:ember: Failed to register group '48689' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:29:08zh:ember: Failed to register group '27905' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:29:10zh:ember: Failed to register group '27921' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:29:12zh:ember: Failed to register group '27937' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:29:14zh:ember: Failed to register group '27953' in multicast table with status=INVALID_STATE.

Even on debug I do not get more logs.
I already added the following to the advanced part of the config:
debounce: 100     # Stabilisiert schnelle Tastenaktionen
  report: false     # Schaltet unnötige Statusmeldungen aus
  suppress_multicast_groups: true  # Unterbindet Multicast-Gruppen
  adapter_options:
    concurrency: 5   # Erhöht die Anzahl der gleichzeitig verwalteten Verbindungen
    retries: 3       # Anzahl der Wiederholversuche bei fehlgeschlagenen Verbindungen
    max_children: 40 # Max. Anzahl an verbundenen Geräten
but it stil gives me error logs

### What did you expect to happen?

That every button press gets reported in the action channel and no error messages.

### How to reproduce it (minimal and precise)

Add at least 8 identical wall switches of that type.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

8.0.2

### Adapter

Sonoff DONGLE-E

### Setup

Pi 4, 4GB, openhabian 64 bit

### Debug log

Error 2024-12-18 09:25:14zh:ember: Failed to register group '48689' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:25:20zh:ember: Failed to register group '2305' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:25:23zh:ember: Failed to register group '2321' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:25:28zh:ember: Failed to register group '2337' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:25:34zh:ember: Failed to register group '2353' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:25:43zh:ember: Failed to register group '24577' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:25:47zh:ember: Failed to register group '24593' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:25:53zh:ember: Failed to register group '24609' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:25:59zh:ember: Failed to register group '24625' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:26:13zh:ember: Failed to register group '27905' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:26:18zh:ember: Failed to register group '27921' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:26:24zh:ember: Failed to register group '27937' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:26:28zh:ember: Failed to register group '27953' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:29:01zh:ember: Failed to register group '48689' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:29:08zh:ember: Failed to register group '27905' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:29:10zh:ember: Failed to register group '27921' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:29:12zh:ember: Failed to register group '27937' in multicast table with status=INVALID_STATE.
Error 2024-12-18 09:29:14zh:ember: Failed to register group '27953' in multicast table with status=INVALID_STATE.

Anpro88, 2024-12-18:
I have to correct my statement slightly, although the Vesternet button is identical in construction, it does not generate any multicast groups when the buttons are pressed and therefore does not generate any errors - could the configuration from the vesternet therefore be used for the ecodim?

Anpro88, 2024-12-18:
Good update, the problem seems to have been solved - I just had to bind every channel of every button to the coordinator, always level ctrl and onoff and it's worked since then.
even if I can't quite understand it...

closed 2024-12-21 by Anpro88

CerielRoland, 2025-04-16:
@Anpro88 how did you do this? i have the same issue with a different rebrand of the button.

Anpro88, 2025-04-16:
Hi @CerielRoland in my case it helped to bind all the endpoints of the device to the coordinator.
If you need further information let me know.
BR Andreas