Koenkk/zigbee2mqtt #25259: No devices and broken settings

patvdleer, 2024-12-18:
### What happened?

All devices are (were) gone from the zigbee2mqtt interface and settings only showed the main and submit button, nothing else. Within settings the other tabs like Advanced, Availibity, etc were gone too.

### How to reproduce it (minimal and precise)

Tools -> Localise device images
restart zigbee2mqtt

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

unrelated

### Adapter

unrelated

### Setup

Add-on on Home Assistant

patvdleer, 2024-12-18:
I fixed it by editing the yaml, `/homeassistant/zigbee2mqtt/configuration.yaml`, and removing all the icons with the base64 line. Saving it and restarting z2m fixes it.

patvdleer, 2024-12-22:
@Grandma-Betty yes, thats what I did, stopped z2m (I don't know if it writes config on shutdown), edited the config by removing all the icon data and started z2m again.

patvdleer, 2024-12-22:
> @patvdleer Thanks, that indeed solved my issues too. But how did you even come to this solution? Did you find anything suspicious within the logs? If so, I think @Koenkk should get aware of this because this seems to be a serious bug. I mean this issue came out of nowhere, I did not change anything and from one day to another that issue occurred.

I remember I pushed the button but didn't update yet from 1.41 to 1.42 and seeing another post that there was an issue with deleted devices in 1.42.0-0 (or 1.42.0-1 IIRC)
 
> Also, just deleting those data lines, who can tell for sure if this will not result in other issues or if the issue will not come back again? I think those lines are there for a reason so we should get to know why this even happened. Just want to be constructive here.

You should be able to test/reproduce it by clicking the `Localise device images` button

Koenkk, 2024-12-22:
Fixing this in https://github.com/Koenkk/zigbee2mqtt/pull/25299

Koenkk, 2024-12-28:
Merged https://github.com/Koenkk/zigbee2mqtt/pull/25299!

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. Note that the dev branch currently contains **breaking changes**, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

closed 2024-12-28 by Koenkk

AncalagonX, 2025-01-06:
@Koenkk Reopening this—I had the same issue, broken settings page with only an empty "Main" tab and a "Submit" button, nothing else. The "Permit join (all)" button still existed, but clicking it did nothing (no timer appeared, no functionality happened).

Here was the broken configuration.yaml:

```
version: 4
homeassistant:
  enabled: true
advanced:
  network_key:
    - 28
    - 154
    - 148
    - 110
    - 249
    - 120
    - 112
    - 210
    - 111
    - 125
    - 7
    - 242
    - 5
    - 193
    - 73
    - 218
  pan_id: 41822
  ext_pan_id:
    - 90
    - 243
    - 82
    - 31
    - 166
    - 210
    - 120
    - 217
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: <longrandompasswordstring>
frontend:
  enabled: true
  port: 8099
```

I deleted the `advanced:` entry and all subordinate entries, restarted Zigbee2MQTT, and suddenly I had all the proper settings tabs.

The worst part about this was that as a new Zigbee2MQTT user, I had no idea it was actually broken in any way. I just thought I didn't understand how to get my devices to appear to Zigbee2MQTT.

jwich71, 2025-01-06:
Same issue after upgrade to 2.0.0. Only Submit button available on Main Settings page. Seems a bug in 2.0.0.
[image]

[image]

AncalagonX, 2025-01-06:
I didn't even have "Advanced / Availability / Blocklist / Frontend / Home Assistant integration / etc." tabs in my settings. Only a blank Main tab and a Submit button.

Koenkk, 2025-01-07:
@AncalagonX do you still have the debug log of this when starting z2m?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

dark-nico, 2025-01-18:
Hi @Koenkk ,
Same issue here, main page is empty after upgrading. 
Everything else seams to work fine. 
I've tried to restart with debug logging but nothing obvious in the logs. 
How can I help ?

Koenkk, 2025-01-19:
If z2m starts fine and just the frontend is not working, it could be that there is something wrong in your reverse proxy config.

dark-nico, 2025-01-19:
I don't use reverse proxy. 
It's only the main settings page that doesn't work, like @jwich71 screenshot. 
Other frontend pages work fine.

noahdroid, 2025-04-15:
Was any solution found to the issue that the main settings page does not work like for @dark-nico and @jwich71 ?
I noticed this today as well