Koenkk/zigbee2mqtt #25260: Adapter EZSP protocol version (8) is not supported by Host [13-14]

Veldkornet, 2024-12-18:
### What happened?

When using the Aeotec Zi-Stick (ZGA008) with the ember driver, I get the following error trying to start z2m:

```
[20:54:36] INFO: Preparing to start...
[20:54:36] INFO: Socat not enabled
[20:54:36] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-12-18 20:54:38] info: 	z2m: Logging to console, file (filename: log.log)
[2024-12-18 20:54:38] info: 	z2m: Starting Zigbee2MQTT version 1.42.0 (commit #unknown)
[2024-12-18 20:54:38] info: 	z2m: Starting zigbee-herdsman (2.1.9)
[2024-12-18 20:54:38] info: 	zh:ember: Using default stack config.
[2024-12-18 20:54:38] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-12-18 20:54:38] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-12-18 20:54:38] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-12-18 20:54:38] info: 	zh:ember:uart:ash: Serial port opened
[2024-12-18 20:54:38] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-12-18 20:54:39] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-12-18 20:54:39] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-12-18 20:54:39] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-12-18 20:54:39] error: 	z2m: Error while starting zigbee-herdsman
[2024-12-18 20:54:39] error: 	z2m: Failed to start zigbee
[2024-12-18 20:54:39] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-12-18 20:54:39] error: 	z2m: Exiting...
[2024-12-18 20:54:40] error: 	z2m: Error: Adapter EZSP protocol version (8) is not supported by Host [13-14].
    at EmberAdapter.emberVersion (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1426:19)
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:688:9)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1598:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)
```

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Aeotec Zi-Stick (ZGA008)

### Setup

Add-on on HomeAssistant OS on Intel NUC

Koenkk, 2024-12-18:
This is expected, update the firmware or switch to the legacy ezsp driver (see the adapter docs)

closed 2024-12-18 by Koenkk

Veldkornet, 2024-12-18:
I already updated to the latest firmware [available](https://aeotec.freshdesk.com/support/solutions/articles/6000270951-update-zi-stick-to-thread-or-zigbee-with-python-3-9-and-windows) from Aeotec, same issue:

```
python Elelabs_EzspFwUtility.py probe -p COM3
2024/12/18 23:10:29 Elelabs_EzspFwUtility:   Elelabs Zigbee adapter detected:
2024/12/18 23:10:29 Elelabs_EzspFwUtility:   Adapter: ELU0142
2024/12/18 23:10:29 Elelabs_EzspFwUtility:   Firmware: 6.10.9-208
2024/12/18 23:10:29 Elelabs_EzspFwUtility:   EZSP v8
```

Koenkk, 2024-12-20:
I'm not sure if a later firmware is available for that device, else you have to stick to the legacy ezsp driver

Veldkornet, 2024-12-21:
FYI, I managed to get a later firmware [here ](https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/aeotec-zga008). I used the ncp-uart-hw-v7.4.5.0-aeotec-zga008-115200.gbl
and flashed it with the [NabuCasa  UniversalSilabsFlasher](https://github.com/NabuCasa/universal-silabs-flasher).
Now I've been able to switch to ember, which is working a lot better. No more errors.

dondersj, 2025-01-17:
i do have the same issues on my ZGA008.
so i want to do the same action as you did.
i have installed the universalsilabsflasher on home assistant.
downloaded the firmware file and put it on a HA folder.
selected the correct USB poort.

i do got the following error in the log:
[15:50:00] INFO: universal-silabs-flasher-up script exited with code 6
s6-rc: warning: unable to start service universal-silabs-flasher: command exited 6
s6-rc: info: service banner: stopping
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.

what do i do wrong? and is there another easy way to put a different FW on my ZGA008?

thanks for your help.

Veldkornet, 2025-01-17:
> i do have the same issues on my ZGA008.
> so i want to do the same action as you did.
> i have installed the universalsilabsflasher on home assistant.
> downloaded the firmware file and put it on a HA [...]

I used the flasher from my Windows PC, if that makes a difference. Not from HA.

dondersj, 2025-01-17:
i also tried, but the software is does not recognise my USB. i do see the usb in windows. it is on com 5

dondersj, 2025-01-17:
at least i tried to use simplicity studio.
is that the correct program to do that?

dondersj, 2025-01-17:
ah. ok. i will try

dondersj, 2025-01-20:
still not managed to get it updated.

not sure if i doing the right steps.

first go to /downloads (there is the *.gbl file)
or do i need to download something more? 

pip install universal-silabs-flasher

universal-silabs-flasher --device COM5 --bootloader-reset yellow flash --firmware ncp-uart-hw-v7.4.5.0-aeotec-zga008-115200.gbl

dondersj, 2025-01-20:
many thanks.
finally i have just found another good link at internet with clear steps which worked.

https://www.broddin.be/upgrading-home-assistant-yellows-zigbee2mqtt-from-ezsp-to-ember/

FlippingBinary, 2025-06-23:
The previous commenter was upgrading their Home Assistant Yellow, which seems to use a different adapter, but I was able to update my Aeotec Zi-Stick model ZGA008 firmware from within the zigbee2mqtt Docker container pretty easily using @Veldkornet's advice. I chose to do it inside the Zigbee2MQTT Docker container because it lets me avoid installing software on the host or plugging the adapter into a different computer. Since the settings are stored on a bind mount, it's trivial to recreate the container from a clean image.

Here's how I updated the firmware of the adapter mounted at `/dev/zigbee`:

1. Open a shell into the docker container `docker exec -it zigbee2mqtt sh`
2. Install Python 3's pip: `apk add py3-pip`
3. Create a virtual environment: `python3 -m venv .venv`
4. Load the virtual environment: `. .venv/source/activate`
5. Install the flasher: `pip install universal-silabs-flasher`
6. Check the current firmware version on the adapter: `universal-silabs-flasher --device /dev/zigbee probe`
7. Download the new firmware if the installed firmware is older than v7.4.5: `wget https://github.com/darkxst/silabs-firmware-builder/raw/refs/heads/main/firmware_builds/aeotec-zga008/ncp-uart-hw-v7.4.5.0-aeotec-zga008-115200.gbl`
8. Install the firmware: `universal-silabs-flasher --device /dev/zigbee flash --firmware ncp-uart-hw-v7.4.5.0-aeotec-zga008-115200.gbl`
9. Check the current firmware version again if you want to double-check that it worked: `universal-silabs-flasher --device /dev/zigbee probe`

I hope this helps anyone else who stumbles on this issue.

cjaca, 2025-10-02:
For me updating **Home Assistant Yellow** internal chip didn't worked today using [https://github.com/mercenaruss/zigstar_addons](https://github.com/mercenaruss/zigstar_addons) addon. 
I had to install flasher in new python venv and update it manually, using modified commands presented by @FlippingBinary.

1. First I checked current state: 
`universal-silabs-flasher --device /dev/ttyAMA1 probe`

[image]

2. Next I downloaded the firmware:

`wget https://github.com/darkxst/silabs-firmware-builder/raw/ember-nohw/firmware_builds/yellow/ncp-uart-hw-v7.4.3.0-yellow-115200.gbl`

3. Flash and check 

`universal-silabs-flasher --device /dev/ttyAMA1 flash --firmware ncp-uart-hw-v7.4.3.0-yellow-115200.gbl`

[image]