Koenkk/zigbee2mqtt #25262: [Feature request]: Device spams network with Info payload resulting from Linkquality Jumps -> Global report limits?

nickeeey, 2024-12-18:
### Is your feature request related to a problem? Please describe

Hi, the Linkquality of the ElectSmart PRO-900Z is jumping and this resulted in a Spam of Information to my network. 

I think a lot of people aren't aware of that and this may result in bad reaction times or crashes on weaker hardware
Thank you!

[image]

### Describe the solution you'd like

I think its maybe helpful to implement a global limit for all devices
(e.g. 1 info per Minute) to reduce the Messages in the Network. I fixed it myself by adding the attribute/ report: Linkquality .... to my yaml (it was not configured in the Interface at the Report tab). 

### Describe alternatives you've considered

Maybe add a information in the documentation to look after that and that the linkquality is a issue which can be fixed by adding the attribute feature

### Additional context

---

nickeeey, 2024-12-18:
little update. I looked up the device log and there is still a info message, so my fix doesnt work

thogens, 2024-12-27:
I seem to have similar issues from time to time. Would be good to:
1: identify the bad guy...
2: control the spamming in some way.

closed 2025-03-05 by github-actions[bot]