Koenkk/zigbee2mqtt #25263: [New device support]: Gas sensor with a supported model, but being recognized as another model

my5123, 2024-12-19:
### Link

https://www.zigbee2mqtt.io/devices/HS1CG.html

### Database entry

{"id":36,"type":"Router","ieeeAddr":"0x8cf681fffeef82f0","nwkAddr":9019,"manufId":4619,"manufName":"HEIMAN","powerSource":"Mains (single phase)","modelId":"GAS-N-ABB","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":53249,"inClusterList":[0,3,23,64929],"outClusterList":[25],"clusters":{"64929":{"attributes":{"0":524288}},"genBasic":{"attributes":{"modelId":"GAS-N-ABB","manufacturerName":"HEIMAN","powerSource":1,"zclVersion":3,"appVersion":17,"stackVersion":2,"hwVersion":16,"dateCode":"2021.12.25"}}},"binds":[{"cluster":25,"type":"endpoint","deviceIeeeAddress":"0x00124b0025e1eae9","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0025e1eae9","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":17,"stackVersion":2,"hwVersion":16,"dateCode":"2021.12.25","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1734583330126}

### Zigbee2MQTT version

1.42.0

### Comments

the model is hs1cg, but being recognized as 'GAS-N-ABB' in z2m.
I used a External definition, but got no message after press the test button on device.

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const reporting = require('zigbee-herdsman-converters/lib/reporting');

const definition = {
    zigbeeModel: ['GAS-N-ABB'],
    model: 'GAS-N-ABB',
    vendor: 'HEIMAN',
    description: 'Automatically generated definition',
    fromZigbee: [fz.ias_gas_alarm_2],
    toZigbee: [],
    exposes: [e.gas(), e.battery_low(), e.tamper()],
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Get no mqtt message
[image]
[image]

my5123, 2024-12-19:
I have try change fromZigbee: [fz.ias_gas_alarm_2] to fromZigbee: [fz.ias_gas_alarm_1], still not working

my5123, 2024-12-19:
Log when press "test" button:

info 2024-12-19 13:36:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"燃气报警器"},"type":"device_announced"}'
info 2024-12-19 13:36:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"燃气报警器","ieee_address":"0x8cf681fffeef82f0"},"type":"device_announce"}'

noirrak, 2025-01-01:
This is the same problem in this device:
https://github.com/Koenkk/zigbee2mqtt/issues/23070

This solution (by @Koenkk) no longer works properly.
https://github.com/Koenkk/zigbee2mqtt/issues/16202#issuecomment-1399431050

my5123, 2025-01-04:
> @my5123, this is the same problem in this device:
> 
> https://github.com/Koenkk/zigbee2mqtt/issues/23070
> 
> 
> 
> This solution (by @Koenkk) no longer works properly:
> 
> https://github.com/Koenkk/zigbee2mqtt/issues/16202#issuecomment-1399431050

Any solutions now?
If u did, could u plz let me known. thx

nonayo, 2025-01-06:
@my5123, did you check the previous solution? This isn't working for me.
https://github.com/Koenkk/zigbee2mqtt/issues/16202#issuecomment-1399431050

my5123, 2025-01-06:
> @my5123, did you check the previous solutions? This isn't working for me.
> 
> https://github.com/Koenkk/zigbee2mqtt/issues/16202#issuecomment-1384405578
> 
> https://github.com/Koenkk/zigbee2mqtt/issues/16202#issuecomment-1399431050

I have tried all the soution u mentioned. It is not work for me too.

closed 2025-03-15 by github-actions[bot]