Koenkk/zigbee2mqtt #25272: Sinope TH1123ZB and TH1123ZB-G2 issues (INSUFFICIENT_SPACE etc.)

reinux, 2024-12-20:
### What happened?

Different things seem to happen with different firmware versions, but the critical issues all seem to come from INSUFFICIENT_SPACE errors, which I can trigger using the Reconfigure button on 1305:

```
error 2024-12-20 00:42:19z2m: Failed to configure 'Upstairs Bathroom Thermostat ', attempt 4 (Error: ZCL command 0x30fb10fffe472575/1 haElectricalMeasurement.configReport([{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":5,"attribute":"activePower"},{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":50,"attribute":"rmsCurrent"},{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":50,"attribute":"rmsVoltage"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'INSUFFICIENT_SPACE') at Endpoint.checkStatus (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28) at Endpoint.zclCommand (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26) at Endpoint.configureReporting (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9) at setupAttributes (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:162:17) at result.configure (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:1950:29) at Object.configure (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/index.ts:205:21) at Configure.configure (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/lib/extension/configure.ts:130:13) at Configure.onMQTTMessage (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/lib/extension/configure.ts:57:21) at EventEmitter.wrappedCallback (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/lib/eventBus.ts:206:17))
```

These devices report the cumulative energy use in watts and so ends up being off by a factor of 1000, and does not update the energy automatically.

Following the discussion at #15988 , I tried setting it to report `divisor` of `seMetering`, but that results in the same error:

```
error 2024-12-20 00:57:15z2m: Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ZCL command 0x30fb10fffe491041/1 seMetering.configReport([{"attribute":"multiplier","minimumReportInterval":60,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'INSUFFICIENT_SPACE')'
```

I have three of these devices, and all behave the same way.

An older device on firmware version 2051 (?) produces this error when attempting to Reconfigure, though it works fine for the most part save for having to manually update activePower:

```
error 2024-12-20 01:00:57z2m: Failed to configure 'Bedroom Thermostat', attempt 1 (Error: ZCL command 0x500b91400002950f/1 haElectricalMeasurement.configReport([{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":5,"attribute":"activePower"},{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":50,"attribute":"rmsCurrent"},{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":50,"attribute":"rmsVoltage"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE') at Endpoint.checkStatus (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28) at Endpoint.zclCommand (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26) at Endpoint.configureReporting (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9) at setupAttributes (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:162:17) at result.configure (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:1950:29) at Object.configure (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/index.ts:205:21) at Configure.configure (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/lib/extension/configure.ts:130:13) at Configure.onMQTTMessage (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/lib/extension/configure.ts:57:21) at EventEmitter.wrappedCallback (/nix/store/awfyawh2p3phc6xzlfsia90xmry7h9p0-zigbee2mqtt-1.41.0/lib/node_modules/zigbee2mqtt/lib/eventBus.ts:206:17))
```

A third device, which I bought a year or so ago and runs firmware version 976, seems to behave fully correctly, and Reconfigure works fine as well.

I'm guessing Z2M has support for a particular version of the firmware (sometime around 976) but not for anything prior or after. I wonder if this can be worked around somehow?

Thanks!

### What did you expect to happen?

Reconfigure completes without errors; all devices report

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

zStack3x0

### Adapter

20240710

### Setup

Running on NixOS on a Surface Go 2 (x86-64)

reinux, 2024-12-26:
Update: All my TH1123ZB-G2 firmware got updated to what the GT130 bridge reports as "0.1.4". The TH1123ZB stays on 0.0.6, but more or less works, so I have no qualms. Version 1305 is the one that produces INSUFFICIENT_SPACE, but presumably, this is the one that needs to be supported?

yanakin1, 2025-01-18:
Same error here. I have 6 thermostat on firmware 1027 that finish configuration and all work well. The 3 other ones I bought later are on firmware 1305 and I get this error. I get similar error as some devices report they are heating at maximum (pi_heating_demand at 100) while they are idle.

reinux, 2025-01-18:
I ended up putting my thermostats on a second coordinator with ZHA.

closed 2025-05-05 by github-actions[bot]

ThatWileyGuy, 2025-08-11:
I have six TH1123ZB thermostats on firmware 2051 that show the same as above:
```
Failed to configure 'Guest Bedroom Thermostat', attempt 4 (Error: ZCL command 0x500b914000029055/1 haElectricalMeasurement.configReport([{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":5,"attribute":"activePower"},{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":50,"attribute":"rmsCurrent"},{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":50,"attribute":"rmsVoltage"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:372:28) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:1005:26) at processTicksAndRejections (node:internal/process/task_queues:105:5) at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:727:9) at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:160:17) at result.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:2090:29) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:21) at Configure.configure (/app/lib/extension/configure.ts:123:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:46:25) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17))
```

Just reading the error on its face, the device is refusing to report one or more of activePower, rmsCurrent, and rmsVoltage, right? How could I go about trying to exclude rmsCurrent or rmsVoltage from the defaults that zigbee2mqtt is requesting?

JoRouss, 2025-10-03:
Did anyone get a fix for this? It's happening to me with 3 TH1123ZB-G2 on firmware 1305